bblanchon/ArduinoJson #42: Ethernet compatability

andrewfasch, 2015-01-04:
Trying to send the JSON data using the Arduino Ethernet library.
Is this possible with the library you have written? 
If so could you provide a small example?

bblanchon, 2015-01-04:
I'll add an example as soon a possible.

I think I'll rewrite the WebServer example provided with the Ardiuno IDE and change the output to be a JSON string instead of a web page.

bblanchon, 2015-01-10:
I added an example using EthernetServer.
Please see [JsonServer.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonServer/JsonServer.ino)

[image]

closed 2015-01-12 by bblanchon