bblanchon/ArduinoJson #43: Energia  (latest version 0014) issue

ilpaso, 2015-01-06:
compiling the IndentedPrintExample the compiler raises:
IndentedPrintExample.cpp.o: In function `setup':
IndentedPrintExample.cpp:(.text.setup+0x16): undefined reference to `vtable for ArduinoJson::Internals::IndentedPrint'

bblanchon, 2015-01-08:
As far as I can tell, it works fine with Arduino.
I'm downloading Energia 0014 and I'll check as soon as I can.

However, by looking at the error, it seems that the file `IndentedPrint.cpp` is not compiled or linked correctly.

bblanchon, 2015-01-12:
I uploaded a package for older versions of the Arduino IDE:

Please give it a try: 
https://github.com/bblanchon/ArduinoJson/releases/download/v4.1/ArduinoJson-v4.1-old-layout.zip

bblanchon, 2015-01-25:
I tested Energia and compared the two packages.
I can now confirm that the "old layout" package fixes the issue.

closed 2015-01-25 by bblanchon