bblanchon/ArduinoJson #44: Non-strict parsing mode in the newest version?

peterpolidoro, 2015-01-09:
You have written a great library, thank you so much! Your code in version 4 looks really clean, I am impressed. It seems that now the JSON parser only operates in what you were calling strict-mode in your previous version. Now if the JSON is not well formed it returns an error instead of parsing in a non-strict manor. Would you consider replacing the original non-strict parsing behavior? Perhaps that would ruin the cleanliness of your new code and it is not reasonable to expect a JSON parser to handle ill formed strings, but I did think it was really nice when your previous code gracefully accepted what now it simply rejects. Previously it seemed that strings inside the JSON did not necessarily need quotation marks, and tokens could be separated with spaces as well as commas. Perhaps even more was allowed through the old non-strict parser that other more strict parsers would reject. I am using your library to parse JSON that I type on a command line and your previous non-strict version saved me from having to type lots of characters that I now need to add. Perhaps I am being lazy and sloppy, but would you please consider restoring the old behavior, at least as an option?

acasadoalonso, 2015-01-09:
I concur …. see my post 
http://forum.arduino.cc/index.php?topic=172578.new#new <http://forum.arduino.cc/index.php?topic=172578.new#new>

Thanks so much for such a great job.
AC/.

bblanchon, 2015-01-10:
Thank you guys :bowtie:
I do make an effort to make the code clear and well tested.

But what I care even more is the code size.
When writing version 4, I first tried to keep that feature, but I finally removed it because it was much simpler without it and allowed more code reuse.

I'll have a look but even if I manage to do it, the code will be significantly bigger so that'll be an optional feature.

peterpolidoro, 2015-01-10:
Perhaps it is better to keep version 4 nice and small. I can just continue to use version 3 for non-strict parsing. Or maybe you can consider adding a non-strict parsing git branch to version 4, so only those who really want that feature have all the extra code. Maybe that is too much for you to have to maintain though. Thanks for looking at it!

acasadoalonso, 2015-01-10:
Benoit,

I do not know if you are aware, but the ArduinoJson library do not compile to either Arduino Galileo or Arduino Edison, with the intel provided Arduino IDE 1.5.3 1.0.0.4. The IDE do not even see the library itself. I think that the problem is related to the libraries property file not being supported by the IDE. If I delete that file the library is seen by the IDE however it  made all kind of compile errors with the code.
I will do a deeper research and let you know

AC/.

bblanchon, 2015-01-10:
No, I didn't know.
The website arduino.cc is down right now, I'll download Arduino 1.5.3 as soon as I can.
Can you please create a new issue?

acasadoalonso, 2015-01-10:
You need to download the IDE from the INTEL site    https://communities.intel.com/docs/DOC-23242 <https://communities.intel.com/docs/DOC-23242>
I will create a new issue on the INTEL forum ….
Cheers.
AC/.

bblanchon, 2015-01-10:
I meant an issue in the ArduinoJson github.

acasadoalonso, 2015-01-12:
I created the issue now on github

bblanchon, 2015-03-01:
I think I'll add a new function to sanitize the JSON string before parsing.
You'll have to call that function yourself, like this:

    char json[64] = "{key:value}";
    sanitizeJson(json, sizeof(json));
    JsonObject& root = jsonBuffer.parseObject();

That way the overhead will only be included if you need the non-strict mode but it would have zero impact for programs that don't need it.

Would that be suitable for you guys?

peterpolidoro, 2015-03-01:
That would be great! Thanks!

bblanchon, 2015-07-10:
Feature available in ArduinoJson 5.0 beta 2.

closed 2015-07-10 by bblanchon