bblanchon/ArduinoJson #45: Library issues with Galileo and Edison IDE from Intel.

acasadoalonso, 2015-01-11:
The current version of the library ArduinoJson V4 is not visible on the MAC when used from the ARDUINO IDE used by Arduino Galileo and Arduino Edison, the IDE provided by INTEL Arduino 1.5.3-Intel 1.0.4                       https://communities.intel.com/docs/DOC-23242
If we delete the file "libraries.properties" from the library, the IDE sees the library, however it has compile errors.
I had no problems when used on the MAC with Arduino IDE 1.5.6-r2
The implementation is really elegant indeed.
I am using it to parse the weather from openweathermap.org on the JSON format API.
Thanks.
AC/.

acasadoalonso, 2015-01-12:
I compiled my test program on my Ubuntu server and runs OK under Arduino IDE 1.5.8 (the latest version available), however it present the same errors under Arduino 1.5.3 Intel 1.0.4
So it is not related to the MAC, it is related to the IDE version.
Thx.
AC/.

bblanchon, 2015-01-12:
OK, I think I found an explanation.

The new library layout is supported since Arduino IDE 1.5.6.
See: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

I'll try to find a way to package the library for older versions.

bblanchon, 2015-01-12:
I uploaded a package for older versions of the Arduino IDE:

Please give it a try: 
https://github.com/bblanchon/ArduinoJson/releases/download/v4.1/ArduinoJson-v4.1-old-layout.zip

acasadoalonso, 2015-01-12:
I tried and it work perfectly, in both Arduino and Edison, thanks so much for your help …. great work indeed.
Cheers.
AC/.

gcartyjr, 2015-01-13:
Amazing! I logged in to send you a message about this, and it's already done! You earned a donation for this! 

Heads up, if you're using the Galileo SD linux image, it contains the "new" header as part of the uclibc library. So you'll have to remove the PlacementNew.hpp from the Arduino Internals or make sure it's not already defined. 

bblanchon feel free to comment.

bblanchon, 2015-01-13:
Thanks a lot! :bowtie: 

This placement new thing is really a pain.
I just can't find a consistent way to declare it when it's missing without conflicting with other declarations.

I think I'll remove it in the next revision.

acasadoalonso, 2015-01-13:
I went thru the same process with the "new" operator.
Reported as an issue and close the issue after discovering the solution.
The original problem is solved with the new layout, I wonder what to do for further releases ???
Thanks for the great work.
AC/.

closed 2015-01-13 by acasadoalonso