bblanchon/ArduinoJson #48: Another virtual destructor issue cropping up

Slechtvalk, 2015-01-16:
Hi,

The pragma you entered suppresses the virtual destructor warning in the static buffer declaration.

I have another issue, this time with the dynamic buffer,  with not declaring the destructor virtual :
`[ 94%] Building CXX object toepassing/usb/CMakeFiles/stm32f3demo.elf.dir/usb_bot.cpp.o
In file in cluded from /Users/oldenburgh/techniek/Variosystem/software/boardtest/torenvalk/../../externals/software/extern/json/ArduinoJson/include/ArduinoJson.h:7:0,
                 from /Users/oldenburgh/techniek/Variosystem/software/boardtest/torenvalk/toepassing/usb/h/usbdemo.h:22,
                 from /Users/oldenburgh/techniek/Variosystem/software/boardtest/torenvalk/toepassing/usb/usb_bot.cpp:38:
/Users/oldenburgh/techniek/Variosystem/software/boardtest/torenvalk/../../externals/software/extern/json/ArduinoJson/include/../include/ArduinoJson/DynamicJsonBuffer.hpp: In destructor 'ArduinoJson::DynamicJsonBuffer::~DynamicJsonBuffer()':
/Users/oldenburgh/techniek/Variosystem/software/boardtest/torenvalk/../../externals/software/extern/json/ArduinoJson/include/../include/ArduinoJson/DynamicJsonBuffer.hpp:20:33: error: deleting object of polymorphic class type 'ArduinoJson::DynamicJsonBuffer' which has non-virtual destructor might cause undefined behaviour [-Werror=delete-non-virtual-dtor]
   ~DynamicJsonBuffer() { delete _next; }
                                 ^
cc1plus: all warnings being treated as errors`

This time, it concerns the dynamic buffer constructor. I see no issue here to declare it virtual, since this class assumes using dynamic allocation anyway.
(The compiler warning shown here originates from arm-none-eabi-g++ 4.8.4 )

bblanchon, 2015-01-16:
It's true that `DynamicJsonBuffer` uses dynamic allocation for its content.

But the class itself is not intended to be dynamically allocated.
Indeed it's designed to avoid that.

You never write `new DynamicJsonBuffer()`.
So you never write `delete myJsonBuffer`.
So the destructor is always statically called.
Therefore it's still not relevant to make it virtual.

And anyway, I bet that if you make it virtual without changing the base class, you'll trigger another warning.

I don't manage to reproduce the warning on older version of the compiler.
Can you try to add a pragma?

bblanchon, 2015-08-20:
I'm sorry I tested on many compilers, including GCC 4.9, but I cannot reproduce this warning.

closed 2015-08-20 by bblanchon

CZEMacLeod, 2018-03-02:
@bblanchon I am getting this issue, and I don't know that I agree with the assertion that you never new the JsonBuffer.
In my case, I am allocating the buffer and storing the pointer and building up a message, perhaps over several function calls or loops.
Once the message is built and written, the buffer needs to be 'released' as you explicitly say don't share or re-use the buffer.
Thus, after saving the json message, I delete the buffer and set the pointer to nullptr.
Before building a new json object
```
	if (jsonBuffer == nullptr) {
		jsonBuffer = new DynamicJsonBuffer();
	}
```
Build the object root
```
        JsonObject & root = (*jsonBuffer).createObject();
```
Later on after finishing with the object root
```
	if (jsonBuffer != nullptr) {
		jsonBuffer->clear();
		delete jsonBuffer;
		jsonBuffer = nullptr;
	}
```
Note that this is done in a class where there can be multiple instances.
Maybe this isn't the best approach, but it certainly seems to work as expected and fulfills my requirements, but if there is another/better way to deal with this I will certainly look at it.
I don't have any memory leaks or other issues and have plenty of space to work with when using a Cortex M4+ type device.

I am not sure the reason you don't have virtual destructors but I found that the compiler warnings (and most documentation say they are actually rather more than warnings) go away if you make the base class have a virtual destructor.
In JsonBufferBase.hpp
```
 protected:
  virtual ~JsonBufferBase() {}
```

bblanchon, 2018-03-07:
Hi @CZEMacLeod,

Why do you use `new` and `delete`?

CZEMacLeod, 2018-03-07:
As I explained in the previous message, I have a process that runs through a number of stages, and adds data to the json message at each stage. Once the message is complete (after many times through loop() and calling to different subroutines) it gets saved to SD and sent and then a new message is started.

The scope of the buffer and root object are global, not local and need to be deleted at the end of each message and newed at the beginning of the next.

bblanchon, 2018-03-08:
I strongly recommend making the buffer local to the loop.
If you believe that this is impossible, just call [JsonBuffer::clear()](https://arduinojson.org/api/jsonbuffer/clear/?utm_source=github&utm_medium=issues)

Regards,
Benoit