bblanchon/ArduinoJson #49: Ver 4.2 may have a memory leak

Tomega3, 2015-02-23:
I modified the JsonParserExample to call a function do do the parsing 
Setup just sets baudrate
loop calls a function that passes in the json char array the parse function runs then delays for 10 seconds
the first 3 times thru the loop it appears all ok, 4 and onward times thru the loop the prse function reports parse Object fail
I can provide test sketch but don't see how to add the attachment to this new issue post

bblanchon, 2015-02-23:
A memory leak is very unlikely if you use `StaticJsonBuffer` because there is no dynamic memory allocation.
However, you need to understand that `StaticJsonBuffer` holds the JSON tokens until it is out of scope, it's the way it is supposed to work.
You probably just have to move the declaration of the `StaticJsonBuffer` inside the loop, so as to reduce the scope of the variable to just were it's needed.
If it doesn't work, please copy/paste the code here.

Tomega3, 2015-02-24:
Hi,

Thanks for the email.

I am attaching my test sketch

/***********************************
Possible memory leak or issue during parse:
Look at the messages on serial monitor which say:
   Serial.print(F("json buffer contents before parse: "));
   Serial.println(json);
   // copy the char array to a temp buffer
   // because it appears that ArduinoJson lib somehow truncates it
   snprintf_P(espbuffer, (sizeof(espbuffer)), PSTR("%s"), json);
   Serial.print(F("json buffer contents in temp buffer before parse: "));
   and
   Serial.print(F("json temp buffer contents after parse: "));
   Serial.println(espbuffer);

   after the parse the espbuffers' contents appear to be truncated.

   also:
   with StaticJsonBuffer<300> jsonBuffer;
   declared as a global variable, the parse reports failure after the 
4th or 5th time thru the loop.

   with it declared inside the parse function ( parseTestDataJSON)
   no reported parse failures are seen but the
   espbuffer is still appears to be truncated.

My concern is that the buffer espbuffer appears to be truncated after 
the parse.

I am running arduino 1.06 on an arduino uno r3

I also tried it on an ESPToy from rayshobbys which has a esp8266 wifi module
that I connect to a weatherdata website to data from
you can find it here::  http://rayshobby.net/?p=10156

On this board my pc appears to slow way down when running the test 
sketch and my GetWifiWeatherData sketch

Thanks again,

Let me know what you find.

Tom

bblanchon, 2015-02-24:
Hi Tom,

Can you please [create a gist](https://gist.github.com/) with the complete `.ino` file?

Also, please have a look at the documentation:

- [Memory model](https://github.com/bblanchon/ArduinoJson/wiki/Memory%20model)
- [Avoiding pitfalls](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls)

Regards,
Benoit

Tomega3, 2015-02-24:
Hi Benoit,

I created a Gist of my test file.
see
https://gist.github.com/Tomega3/2f9e208f61da69edafbf

I have read the memory model and the avoiding pitfalls

I hope this helps

Thanks again,

Tom

bblanchon, 2015-02-24:
I'm sorry, I think I overlooked your sentence:

> My concern is that the buffer espbuffer appears to be truncated after
the parse.

This is an expected behavior, please read carefully [Avoiding pitfalls, section 6](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#6-make-sure-the-string-isnt-read-only):

> Indeed, the parser modifies the string for two reasons:
> - it inserts \0 to terminate substrings,
> - it translate escaped characters like \n or \t.

This is how the library works.
It's done this way to avoid memory copy.
If you do need a copy of the original string, you need to make a copy in your program.

Tomega3, 2015-02-24:
Thank you very much for the info.

If you want to close and or remove the issue I posted please do so.

Thanks again,

Tom

closed 2015-02-24 by bblanchon