bblanchon/ArduinoJson #50: Get JSON String Length in Arduino

tombarker1804, 2015-02-25:
I want to send a JSON string via UDP from my Arduino. I was thinking to use this method (from the wiki):

char buffer[256];
array.printTo(buffer, sizeof(buffer));

I would prefer not to allocate such a large buffer. Is there a way to get the length of the string that will be generated so the buffer can be sized accordingly?

bblanchon, 2015-02-25:
You don't need to use a buffer, you can directly print the JSON object to `Udp`.

    Udp.beginPacket(remoteIP, port);
    root.printTo(Udp);
    Udp.endPacket();

Please see the new example: [JsonUdpBeacon](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonUdpBeacon/JsonUdpBeacon.ino)

tombarks, 2015-02-25:
Awesome! Thanks so much for this help and this great library!

Tom

closed 2015-02-26 by bblanchon