bblanchon/ArduinoJson #51: nested Array can not contain other arrays?

lazyzero, 2015-02-26:
Hi,

I like to do something like 

 JsonArray& data = root.createNestedArray("data");
  
  JsonArray& data1 = data.createNestedArray("data1");
  data1.add(48.756080, 6); 
  data1.add(2.302038, 6); 
  JsonArray& data2 = data.createNestedArray("data2");
  data2.add(48.756080, 6);
  data2.add(2.302038, 6);

  data.add(data1);
  data.add(data2);

But get this compilation error
JsonGeneratorExample.ino:19:52: error: no matching function for call to 'ArduinoJson::JsonArray::createNestedArray(const char [6])'
JsonGeneratorExample.ino:19:52: note: candidate is:
In file included from /home/cccc/Arduino_sketchbook/libraries/ArduinoJson/include/ArduinoJson.h:8:0,
                 from /home/cccc/Arduino_sketchbook/libraries/ArduinoJson/ArduinoJson.h:11,
                 from JsonGeneratorExample.ino:7:
/home/cccc/Arduino_sketchbook/libraries/ArduinoJson/include/../include/ArduinoJson/JsonArray.hpp:68:14: note: ArduinoJson::JsonArray& ArduinoJson::JsonArray::createNestedArray()
   JsonArray &createNestedArray();

My JSON string should look like this {"sensor":"gps","time":1351824120,"data":[[48.756080,2.302038],[48.756080,2.302038]]}

Isn't this possible or where is my error?

Best Christian

bblanchon, 2015-02-26:
Hi Christian,

Signatures of `createNestedArray()` are different for `JsonArray` and `JsonObject`:

    JsonArray& JsonArray::createNestedArray();    
    JsonArray& JsonObject::createNestedObject(const char* key);

So, you should do:

    JsonArray& data = root.createNestedArray("data");

    JsonArray& data1 = data.createNestedArray();
    data1.add(48.756080, 6); 
    data1.add(2.302038, 6); 

    JsonArray& data2 = data.createNestedArray();
    data2.add(48.756080, 6);
    data2.add(2.302038, 6);

Regards,
Benoit

lazyzero, 2015-02-27:
Thanks for your assistance. Works now.

Christian

closed 2015-02-27 by lazyzero

pkourany, 2016-02-17:
@bblanchon, is it possible to create an array of arrays with a variable name for the nested array.  I want to encode a variable number of 3-value int data points (nested array) within an array.  Using the approach above means hard-coding each nested array name which won't work in my case.

Thanks!
Paul

pkourany, 2016-02-17:
I may have found the solution in closed issues.  It compiles but not sure if it works:
```
	JsonArray& data = object.createNestedArray("data");

	for (int i = 0; i < iLength_of_list; i++) {
		JsonArray &values = data.createNestedArray();
		values.add(Time.local());
		values.add(refV[i]);
		values.add(gauge[i]);
	}
```
It's not an array of arrays but an object of arrays which will do the trick if it works. 

Thanks!

bblanchon, 2016-02-17:
@pkourany you're code looks ok, it should work fine

pkourany, 2016-02-17:
Great!!! I'll test tonight.  And thanks for a great library +1:

gha2al, 2018-06-28:
hi
im sorry it's really emergency 
if any ones  know this please help me
my case is almost similar
this is my code:

#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <SimpleDHT.h>

int pinDHT11 = 2;
SimpleDHT11 dht11;

int ledPin=5;

void setup() {

 byte temperature = 0;
  byte humidity = 0;
  pinMode(ledPin, OUTPUT);

  Serial.begin(115200);                            //Serial connection
  WiFi.begin("iPhone", "kaz2y5:D");   //WiFi connection

  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion

    delay(500);
    Serial.println("Waiting for connection");

  }

}
bool toggle=false;

void loop() {

  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status

    DynamicJsonBuffer JSONbuffer;
    JsonObject& JSONencoder = JSONbuffer.createObject(); 

    

    JsonArray& t = JSONencoder.createNestedArray(); //JSON array
    JsonArray& tvlue = t.createArray("value"); //JSON array
    JsonArray& ttime = t.createArray("time"); //JSON array
    JsonArray& h = JSONencoder.createNestedArray(); //JSON array
    JsonArray& hvalue = h.createArray("value"); //JSON array
    JsonArray& htime = h.createArray("time"); //JSON array

    byte temperature;
    byte humidity;
    int err = SimpleDHTErrSuccess;
    if ((err = dht11.read(pinDHT11, &temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {
        Serial.print("Read DHT11 failed, err="); Serial.println(err);delay(1000);
        return;
      }

    tvalue["value"]=((int)temperature); //Add value to array
    ttime["time"]=(int(millis())); //Add value to array

    hvalue["value"]=((int)humidity); //Add value to array
    htime["time"]=(int(millis())); //Add value to array

    char JSONmessageBuffer[300];
    JSONencoder.prettyPrintTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
    Serial.println(JSONmessageBuffer);

    HTTPClient http;    //Declare object of class HTTPClient
[... 207 lines cut ...]

                 from C:\Users\sony\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from E:\iot\Final\4\wemos\q4\q4.ino:3:

C:\Users\sony\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonArraySubscript.hpp:110:38: error:   initializing argument 1 of 'ArduinoJson::Internals::JsonArraySubscript ArduinoJson::JsonArray::operator[](size_t)' [-fpermissive]

 inline Internals::JsonArraySubscript JsonArray::operator[](size_t index) {

                                      ^

Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\sony\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi
 Not used: E:\Narm Afzar\arduino 1.8.5\libraries\WiFi
 Not used: E:\Narm Afzar\arduino 1.8.5\libraries\WiFi
 Not used: E:\Narm Afzar\arduino 1.8.5\libraries\WiFi
 Not used: E:\Narm Afzar\arduino 1.8.5\libraries\WiFi
exit status 1
no matching function for call to 'ArduinoJson::JsonObject::createNestedArray()'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

what should I do?