bblanchon/ArduinoJson #52: 4.2 wont verify in eclipseArduinoIDE

yuanhaoliang, 2015-03-02:
I download the eclipseArduinoIDE from http://www.baeyens.it/eclipse/,

When I verify JsonParserExample.ino  in the eclipseArduinoIDE using ArduinoIDE1.5.6-r2 which eclipse insists to,
I get the error:

Starting combiner
"E:/arduino/LinkItOne/arduino-1.5.6-r2/hardware/tools/g++_arm_none_eabi/bin/arm-none-eabi-g++" -O2 -Wl,--gc-sections -mcpu=arm7tdmi-s "-T/E:/arduino/LinkItOne/arduino-1.5.6-r2/hardware/arduino/mtk/variants/linkit_one/linker_scripts/gcc/scat.ld" "-Wl,-Map,E:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release/Test.map" -o "E:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release/Test.elf" "-LE:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release" -lm -fpic -msvr4-struct-return -pie -Wl,--entry=gcc_entry -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-unresolved-symbols -Wl,--start-group "E:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release/arduino/syscalls_mtk.c.o"    ./.ino.cpp.o  ./Libraries/Wire/example/LiquidCrystal_I2C1602V1/LiquidCrystal_I2C.cpp.o  ./Libraries/Wire/Wire.cpp.o  ./Libraries/Stepper/Stepper.cpp.o  ./Libraries/Servo/Servo.cpp.o  ./Libraries/SPI/SPI.cpp.o  ./Libraries/LWiFi/LWiFi.cpp.o ./Libraries/LWiFi/LWiFiClient.cpp.o ./Libraries/LWiFi/LWiFiServer.cpp.o ./Libraries/LWiFi/LWiFiUdp.cpp.o  ./Libraries/LStorage/LFlash.cpp.o ./Libraries/LStorage/LSD.cpp.o ./Libraries/LStorage/LStorage.cpp.o  ./Libraries/LGSM/utility/VoiceCall.cpp.o  ./Libraries/LGSM/LSMS.cpp.o ./Libraries/LGSM/LVoiceCall.cpp.o  ./Libraries/LGPS/utility/GPS.cpp.o  ./Libraries/LGPS/LGPS.cpp.o  ./Libraries/LGPRS/LGPRS.cpp.o ./Libraries/LGPRS/LGPRSClient.cpp.o ./Libraries/LGPRS/LGPRSServer.cpp.o ./Libraries/LGPRS/LGPRSUdp.cpp.o  ./Libraries/LGATT/utility/GATTClient.cpp.o ./Libraries/LGATT/utility/GATTServer.cpp.o  ./Libraries/LGATT/LGATTClient.cpp.o ./Libraries/LGATT/LGATTServer.cpp.o ./Libraries/LGATT/LGATTUUID.cpp.o  ./Libraries/LEEPROM/utility/sysfile.cpp.o  ./Libraries/LEEPROM/LEEPROM.cpp.o  ./Libraries/LDateTime/LDateTime.cpp.o  ./Libraries/LBattery/utility/Battery.cpp.o  ./Libraries/LBattery/LBattery.cpp.o  ./Libraries/LBT/utility/BTClient.cpp.o ./Libraries/LBT/utility/BTServer.cpp.o  ./Libraries/LBT/LBT.cpp.o ./Libraries/LBT/LBTClient.cpp.o ./Libraries/LBT/LBTServer.cpp.o  ./Libraries/LAudio/utility/Audio.cpp.o  ./Libraries/LAudio/LAudio.cpp.o  ./Libraries/ArduinoJson/src/Internals/IndentedPrint.cpp.o ./Libraries/ArduinoJson/src/Internals/JsonParser.cpp.o ./Libraries/ArduinoJson/src/Internals/List.cpp.o ./Libraries/ArduinoJson/src/Internals/Prettyfier.cpp.o ./Libraries/ArduinoJson/src/Internals/QuotedString.cpp.o ./Libraries/ArduinoJson/src/Internals/StringBuilder.cpp.o  ./Libraries/ArduinoJson/src/Arduino/Print.cpp.o  ./Libraries/ArduinoJson/src/JsonArray.cpp.o ./Libraries/ArduinoJson/src/JsonBuffer.cpp.o ./Libraries/ArduinoJson/src/JsonObject.cpp.o ./Libraries/ArduinoJson/src/JsonVariant.cpp.o  ./Libraries/ArduinoJson/ArduinoJson.cpp.o   E:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release/arduino.ar   "/E:/arduino/LinkItOne/arduino-1.5.6-r2/hardware/arduino/mtk/variants/linkit_one/libmtk.a" "E:/arduino/LinkItOne/eclipseArduino/workspace/Test/Release/arduino.ar" -Wl,--end-group
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::Internals::IndentedPrint::write(unsigned char)':
E:/arduino/workspace/libraries/ArduinoJson/src/Internals/IndentedPrint.cpp:11: multiple definition of `ArduinoJson::Internals::IndentedPrint::write(unsigned char)'
./Libraries/ArduinoJson/src/Internals/IndentedPrint.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/Internals/IndentedPrint.cpp:11: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::Internals::JsonParser::skip(char const*)':
E:/arduino/workspace/libraries/ArduinoJson/src/Internals/JsonParser.cpp:32: multiple definition of `ArduinoJson::Internals::JsonParser::skip(char const*)'
./Libraries/ArduinoJson/src/Internals/JsonParser.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/Internals/JsonParser.cpp:32: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::Internals::QuotedString::extractFrom(char*, char**)':
E:/arduino/workspace/libraries/ArduinoJson/src/Internals/QuotedString.cpp:60: multiple definition of `ArduinoJson::Internals::QuotedString::extractFrom(char*, char**)'
./Libraries/ArduinoJson/src/Internals/QuotedString.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/Internals/QuotedString.cpp:60: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::Internals::StringBuilder::write(unsigned char)':
E:/arduino/workspace/libraries/ArduinoJson/src/Internals/StringBuilder.cpp:11: multiple definition of `ArduinoJson::Internals::StringBuilder::write(unsigned char)'
./Libraries/ArduinoJson/src/Internals/StringBuilder.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/Internals/StringBuilder.cpp:11: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonArray::at(int) const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonArray.cpp:17: multiple definition of `ArduinoJson::JsonArray::at(int) const'
./Libraries/ArduinoJson/src/JsonArray.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonArray.cpp:17: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonBuffer::createArray()':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonBuffer.cpp:16: multiple definition of `ArduinoJson::JsonBuffer::createArray()'
./Libraries/ArduinoJson/src/JsonBuffer.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonBuffer.cpp:16: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonBuffer::createObject()':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonBuffer.cpp:21: multiple definition of `ArduinoJson::JsonBuffer::createObject()'
./Libraries/ArduinoJson/src/JsonBuffer.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonBuffer.cpp:21: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator ArduinoJson::JsonArray&() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:17: multiple definition of `ArduinoJson::JsonVariant::operator ArduinoJson::JsonArray&() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:17: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator ArduinoJson::JsonObject&() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:21: multiple definition of `ArduinoJson::JsonVariant::operator ArduinoJson::JsonObject&() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:21: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator bool() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:25: multiple definition of `ArduinoJson::JsonVariant::operator bool() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:25: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator char const*() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:29: multiple definition of `ArduinoJson::JsonVariant::operator char const*() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:29: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator double() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:33: multiple definition of `ArduinoJson::JsonVariant::operator double() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:33: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::operator long() const':
E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:37: multiple definition of `ArduinoJson::JsonVariant::operator long() const'
./Libraries/ArduinoJson/src/JsonVariant.cpp.o:E:/arduino/workspace/libraries/ArduinoJson/src/JsonVariant.cpp:37: first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o: In function `ArduinoJson::JsonVariant::set(bool)':
[... cut ...]
./Libraries/ArduinoJson/ArduinoJson.cpp.o:(.data.rel.ro+0x10): multiple definition of `vtable for ArduinoJson::Internals::Prettyfier'
./Libraries/ArduinoJson/src/Internals/Prettyfier.cpp.o:(.data.rel.ro+0x0): first defined here
./Libraries/ArduinoJson/ArduinoJson.cpp.o:(.data.rel.ro+0x20): multiple definition of `vtable for ArduinoJson::Internals::IndentedPrint'
./Libraries/ArduinoJson/src/Internals/IndentedPrint.cpp.o:(.data.rel.ro+0x0): first defined here
collect2: ld returned 1 exit status

Only the ArduinoJson will be error. help.....

bblanchon, 2015-03-02:
You probably just need to delete (or exclude) `ArduinoJson.cpp` at the root of the library.
This file is including all other `.cpp` files, this is just for the classic Arduino IDE.

You can see I already put a `#ifdef ARDUINO` in this file, in hope it would be ignores by other IDEs.
But apparently it doesn't work.

I would be very interested if you find another preprocessor symbol that we could use to exclude this file on Eclipse.

yuanhaoliang, 2015-03-02:
OK, I will try.

dcsoftware, 2015-03-06:
I had the same problem and resolved it by excluding the "src" folder from building.
After you include the library in your project, expand Libraries->ArduinoJson->right click on the "src" folder->select Resource Configurations->Exclude from Build-> tick the release option. OK and verify again, it should work now!

closed 2015-03-12 by bblanchon