bblanchon/ArduinoJson #54: Parsing Help for JSON / Arduino NOOB!

alxscott, 2015-03-04:
Firstly, apologies for raising an "Issue", it is certainly not an issue as such, I'm just after a gentle nudge in the right direction!

I'm using the NinjaCloud platform to try and control a RGB strip for my final engineering project. NinjaCloud uses Json / REST to communicate, and your Wiki and library has made basic Json parsing incredibly clear; so thank you for that!

My problem is the JSON string I need to parse:

` {"DEVICE":[{"G":"0","V":0,"D":1011,"DA":"{\"on\":true,\"hue\":65535,\"bri\":222}"}]} `

The information I want to parse is the "On", "Hue" and "Bri(ghtness)" elements; but having tried to use this string against your JSON Parse example, I can't get my head around if the "DA" part is an array, a child, a sibling... As I said I'm very new to JSON so very confused!

My understanding goes as far as the root is "DEVICE", which is an array containing "G", "V" "D" labels and associated values.

That said, in the short time I've been trying to learn JSON, your library is infinitely simpler than any of the others I've tried to use to understand JSON!

Any words of advice would be very much appreciated, and thanks again!

Alex

bblanchon, 2015-03-04:
[image]

The solution is to parse twice: once for the whole string and another one for the nested JSON in the "DA" member:

    JsonObject& root = jsonBuffer.parseObject(json);

    char buffer[128];
    strcpy(buffer, root["DEVICE"][0]["DA"]);
    JsonObject& da = jsonBuffer.parseObject(buffer);

    bool on = da["on"];
    long hue = da["hue"];
    long bri = da["bri"];

See [Issue54.cpp](https://github.com/bblanchon/ArduinoJson/blob/issue54/test/Issue54.cpp).

alxscott, 2015-03-04:
Thats awesome!! Thanks a lot for that; and the worked example!

Very much appreciated!!!

alxscott, 2015-03-04:
Well, this is slightly embarrassing... I went back to my project, and discovered the Json string is "split" by the NinjaBlock ethernet library already :/. However, after thinking my life had got way simpler, I'm still not managing to parse the returned string :(.

This is my sketch:

```C

#include <SPI.h>
#include <Ethernet.h>
#include <NinjaBlockEthernet.h>
#include <ArduinoJson.h>
//#include <MemoryFree.h>

#define DEFAULT_VENDOR_ID 0
#define LED_DEVICE_ID 1011

#define ENABLE_SERIAL true    // @justy : You may wish to disable Serial, but it's also handy to be able to use it for debugging, so here's a master switch.

byte led = 6;  // Connect the anode (long lead, +ve) of a LED to this pin, and connect that LED's cathode (short lead, -ve) to GND through a 330R-1K resistor.

void setup() {
  pinMode(led, OUTPUT);
#if ENABLE_SERIAL
  Serial.begin(9600);
  Serial.println("Starting..");
#endif
  delay(1000);   // This delay is to wait for the Ethernet Controller to get ready

  NinjaBlock.host = "api.ninja.is";
  NinjaBlock.port = 80;
  NinjaBlock.nodeID = "ETHERSHIELDBLOCK";  // Name this as you wish
  NinjaBlock.token = "c969dd1dc483ae661853b3a408f3fbfe2f9cdd40"; // Get yours from https://a.ninja.is/hacking
  NinjaBlock.guid = "0";
  NinjaBlock.vendorID = DEFAULT_VENDOR_ID;
  NinjaBlock.deviceID = LED_DEVICE_ID;

  if (NinjaBlock.begin() == 0)
    Serial.println("Init failed");

  // Tell Ninja we exist and are alive and off.
#if ENABLE_SERIAL
  Serial.println("Creating LED");
#endif
  NinjaBlock.send("000000"); // We send 000000 because we are identifying as an RGB led

}

void loop() {

  if (NinjaBlock.receive()) {
    // If this function returns true, there are either commands (data) from the server
    // or a keep alive Tick has been sent
    //First, you MUST check if it is a Tick
    if (NinjaBlock.IsTick) { //There is no input, just a keep alive update

      //Make sure any devices tha don't regularly send data to the cloud
      // are updated here or they will time out

    } else { //There is data to respond to

      // Return values are:
      // NinjaBlock.strGUID
      // NinjaBlock.intVID
      // NinjaBlock.intDID
      // NinjaBlock.intDATA - if data is integer
      // NinjaBlock.strDATA - if data is string (note char[64])

      if (NinjaBlock.IsDATAString) {

        Serial.print("strDATA=");
        Serial.println(NinjaBlock.strDATA);

        StaticJsonBuffer<50> jsonBuffer;

        //char json[]="{\"on\":true,\"hue\":30537,\"bri\":52}";

        JsonObject& root = jsonBuffer.parseObject(NinjaBlock.strDATA);//parse JSON String for NinjaCloud

        if (!root.success()) {
          Serial.println("parseObject() failed");
          return;
        }

        const char* state = root["on"];
        long hue = root["hue"];
        long bri = root["bri"];

        Serial.println(state);
        Serial.println(hue);
        Serial.println(bri);

      }
    }
  }
}
```

This produces the following serial output:
```Starting..
IP: 192.168.1.99.
Creating LED
_Sent={"GUID": "ETHERSHIELDBLOCK_0_0_1011","G": "0","V": 0,"D": 1011,"DA": "000000"}
...._Recv=55
{"DEVICE":[{"G":"0","V":0,"D":1011,"DA":"{\"on\":true,\"hue\":35684,\"bri\":190}"}]}

0

strDATA={\"on\":true,\"hue\":35684,\"bri\":190}
parseObject() failed
```

If I use the same code, but copy the NinjaBlock.strDATA from the serial output, insert into the code and parse it it works as it should (i.e. Doesn't reflect the new string, but parses the serial output as it should:

```C
      if (NinjaBlock.IsDATAString) {

        Serial.print("strDATA=");
        Serial.println(NinjaBlock.strDATA);

        StaticJsonBuffer<50> jsonBuffer;

        char json[]="{\"on\":true,\"hue\":30537,\"bri\":52}";

        JsonObject& root = jsonBuffer.parseObject(json);//parse JSON String for NinjaCloud

        if (!root.success()) {
          Serial.println("parseObject() failed");
          return;
        }
```

Serial Output:
```
Starting..
IP: 192.168.1.99.
Creating LED
_Sent={"GUID": "ETHERSHIELDBLOCK_0_0_1011","G": "0","V": 0,"D": 1011,"DA": "000000"}
..._Recv=54
{"DEVICE":[{"G":"0","V":0,"D":1011,"DA":"{\"on\":true,\"hue\":27792,\"bri\":27}"}]}

strDATA={\"on\":true,\"hue\":27792,\"bri\":27}

30537
52
```

It says in your Wiki that there could be three reasons why it fails. I've checked the string length, so I know the buffer is sufficient; I'm confident it is an object, and not an array (Tho I'm only getting used to the ideas). Finally, it is invalid. My only thought is could it be something to do with escaped characters again.

Is there any ways to 'fault find' the raw Json to see where the error lies?!

Many Thanks (Again!)

Alex

bblanchon, 2015-03-04:
Thanks for the donations!

I don't have a quick solution for you second problem because `NinjaBlock.strDATA` doesn't contain valid JSON.

I'm really concerned about this line on the serial output:

    strDATA={\"on\":true,\"hue\":27792,\"bri\":27}

Quote characters are escaped but they should not.

You don't see the issue when you define the variable in your program because you set it wrong. To match your input it should be:

    char json[]="{\\\"on\\\":true,\\\"hue\\\":30537,\\\"bri\\\":52}";

Here are two dirty workarounds:

1. remove the backslash from  `NinjaBlock.strDATA`
2. fool the JSON parser by nesting `NinjaBlock.strDATA` in a string value (`sprintf(buffer, "[\"%s\"]", NinjaBlock.strDATA)` for instance

Of course the best would be to get `NinjaBlock.strDATA` in the right format, but I assume you can't change that.

alxscott, 2015-03-07:
Thanks a lot for that, and your very welcome for the donation... I'd been searching for days to try and get my head around it!

Still haven't managed to get my sketch to properly parse; hence the delays reply :).

bblanchon, 2015-03-07:
Well, **it's not your fault**, the data you receive in `NinjaBlock.strDATA` is very bad.
There is no easy way to parse that string.
All the solution that I can imagine involve a lot of ugly and inefficient code, which is not an option on a tiny platform like Arduino.

If I were you, i would try to replace the library `NinjaBlockEthernet` with my own implementation.
Making a HTTP request and parsing JSON is pretty easy.
Have a look at my [RFID terminal](https://github.com/drinks-wallet/DrinksRfidTerminal) for an example.

alxscott, 2015-03-07:
Thanks for all your help!

I decided to come at it from a completely different angle; and treat it as a simple string; not a Json entity. Just incase it helps anybody on Google; this is my code... Its not very pretty nor efficient; but it extracts the values I need and should get my project finished on time :)

```C
      if (NinjaBlock.IsDATAString) {

        Serial.print("strDATA=");
        Serial.println(NinjaBlock.strDATA);
        // char stringToParse[] = "{\\\"on\\\":true,\\\"hue\\\":30537,\\\"bri\\\":52}";
        const char *delim  = ": ,";   //a comma is the delimiter

        char *trim1;
        char *on;
        char *trim2;
        char *hue;
        char *trim3;
        char *bri;
        char *trim4;

        trim1 = strtok(NinjaBlock.strDATA, delim);
        on = strtok(NULL, delim);
        trim2 = strtok(NULL, delim);
        hue = strtok(NULL, delim);
        trim3 = strtok(NULL, delim);
        bri = strtok(NULL, "}");

        Serial.print ("On: ");
        Serial.println(on);
        Serial.print ("Hue: ");
        Serial.println(hue);
        Serial.print ("Brightness: ");
        Serial.println(bri);
        int brightness = atoi(bri);
        char buf[8]; //Using a buffer for String->char[]
        itoa(brightness, buf, 10);
        analogWrite(R_LED, brightness);
```

Thanks again for your help and guidance, very much appreciated!

Alex

closed 2015-03-12 by bblanchon