bblanchon/ArduinoJson #55: Assign String to key doesn't work

dcsoftware, 2015-03-06:
Hi, I run into this issue and I don't know if I am doing something wrong or there is a problem in the library.
When I assign a string value to a key in an object, what it is assigned is true value. I have a string received from Serial witch I cut into several substring and them create a json with this data.
Here is a sample of my code:

char buff[64];

Serial.readBytesUntil(10, buff, sizeof(buff));

String s = buff;

StaticJsonBuffer<300> jsonBuffer;

JsonObject& root = jsonBuffer.createObject();

root["key"] = s.substring(10, 20);
root["key1"] = s.substring(21, 28);

root.printTo(Serial);

The output is:
{"key":true,"key1":true}

instead of having my substring it store a "true" value.
Same thing happens if I do this:

String a = s.substring(10,20);

root["key"] = a;

Waiting for an answer a in the mean time, Thank very much for your awesome work!!!

bblanchon, 2015-03-06:
The current version of Arduino JSON doesn't directly support the `String` class.
Right now the work around is to convert the `String` to a `char*` by calling `.c_str()`.

For instance:

    root["key"] = s.substring(10, 20).c_str();

closed 2015-03-07 by bblanchon

arnold-b, 2018-02-24:
Dont work at all
ConfigClass::setparam(String name,String param, String val)
{
	if (sysconf_loaded == -1)
	{
		loadconfigfile(0);
	}
	if (sysconf_loaded != -1)
	{
		Serial.println("setparam " + name + " " + param + " = " + val);
		DynamicJsonBuffer jsonBuffer(1024);
		JsonObject& root = jsonBuffer.parseObject(cfg);
		String temp = "";
		if (param.length() > 0)
		{
			Serial.println("Setparam root[name][param]");
			temp = root[name.c_str()][param.c_str()].as<String>();
		}
		else 
		{
			Serial.println("Setparam root[name]");
			temp = root[name.c_str()].as<String>();
		}

		if (temp != val)
		{
			Serial.println("Setparam New");
			need2save = true;
			root[name.c_str()][param.c_str()] = val.c_str();
			
			cfg = "";
			root.printTo(cfg);
			root.printTo(Serial);
		}
i just get {} as result

bblanchon, 2018-02-26:
Hi @arnold-b,

The information on this thread is obsolete.
You now need to remove `.c_str()`.
If you need more help, please open a new issue.

Regards,
Benoit