bblanchon/ArduinoJson #57: Encoding variable number of nested objects

ChrisKal, 2015-03-10:
Hello and thank you for your great work.
I use your library for building a multisensor platform (based on an Arduino Mega) for indoor air quality and use JSON for communicating with an handheld device. Your library does quite a job in my project. However I still have one problem for which i just can't find a proper solution.

My idea is generate a variable number of nested object wihtin an iteration. In this iteration I try to assign the values of an array of integers and an corresponding array of chararrays to the key-value pairs of each object. I do understand, that I can't build an array of objects, since the objects are references theirself. But I don't know how to deal with that probelm.

Here is what i want to do in pseudo code:
 
```C++
int iLength_of_list
int list[3];
list[0] = test1;
list[1] = test2;
list[2] = test3;

StaticJsonBuffer<256> jsonBuffer;
JsonObject& joMessage = jsonBuffer.createObject();
joMessage["length of list"] = iLength_of_list;

/* some magic must happen here, similar to JsonObject& nestedObject[iLength_of_list] */

for (int i = 0; i<= iLength_of_list-1;i++) {
    char cBuffer[2];
    sprintf(cBuffer, "%d", i)
    nestedObject[i] = joMessage.createNestedObject(cBuffer);
    nestedObject[i]["en"] = englishTextOfIndex(list[i]);
    nestedObject[i]["de"] = germanTextOfIndex(list[i]);
}
joMessage.printTo(Serial)
```

I hope you can get an idea of what I am trying to do. I have read through all the issues and think that franklinvv in [issue #10]  (https://github.com/bblanchon/ArduinoJson/issues/10) had a similar problem, but the examplecode does not run for me.

Many thanks in advance

Chris

bblanchon, 2015-03-10:
Hi Chris,

The problem is the local variable `cBuffer`.
By using this variable as a name in the JSON object tree, you're implicitly pointing to a variable that doesn't exists outside of the loop.

The solution really depends on your project.

The simpler would be to change `joMessage` into a `JsonArray`instead of a `JsonObject`, so as to avoid naming the nested elements.

If you absolutely need to name the elements, you'll need to store the names in a variable that is outside of the loop scope, for instance:

    char names[20][2];

    for(...) {
        sprintf(names[i], "%d", i);
        ...
    }

Regards,
Benoit

ChrisKal, 2015-03-12:
Thank you for your quick response.

Unfortunately, i can't use a `JsonArray` for this task. I absolutely need to name the elements because there already exists a definition for the serial communication interface I try to develop on the Arduino.
The final result should look like this:
```
"List": {
	"Length": 5,
	"0": {
		"en": "dog",
		"de": "Hund"
	}, 
	"1": {
		"en": "cat",
		"de": "Katze"
	}, ...
	"2": {
		"en": "mouse",
		"de": "Maus"
	}, ...
}
```

>If you absolutely need to name the elements, you'll need to store the names in a variable that is outside of the loop scope, for instance:

 Your are right here, my examplecode is wrong. But that is not the actual problem. I try to generate an Array with a variable number of JSON elements. My aim ist to iterate throguh them:

```c++
int iLength_of_list = 5;
...
char * buff_ListItem;
buff_ListItem =  new char[3];
JsonObject& mainObject;
JsonObject&* nestedObject;
for (int i = 0; i<iLength_of_list-1; i++){
    sprintf(buff_ListItem,"%d",i);
    nestedObject[i] = jsonBuffer.createObject();
    nestedObject[i]["en"] = english[i];
    nestedObject[i]["de"] = german[i];
    mainObject.add(buff_ListItem, nestedObject[i]);
}
delete[] buff_ListItem;
```
As far as I do understand, i can't create an array of `JsonObject&` and my problem that it does not have a default constructor.

Thank you for your help,

Chris

bblanchon, 2015-03-12:
Please try:

    int iLength_of_list = 5;
    ...
    char itemNames[MAX_COUNT][2];
    for (int i = 0; i<iLength_of_list; i++) {
        sprintf(itemNames[i],"%d",i);
        JsonObject& nestedObject = mainObject.createNestedObject(itemNames[i]);
        nestedObject["en"] = english[i];
        nestedObject["de"] = german[i];
    }

closed 2015-04-16 by bblanchon

markusand, 2015-07-01:
Hi! Related to this topic, I'm trying to implement a JSON nested chain similar to this:
```
{
  "id" : "398274",
  "device" : "Ambient+",
  "version" : "1.2",
  "data" : [{
    "temperature" : "22.5",
    "humidity" : "83",
    "lux" : "4208",
    "CO2" : "465"
  }]
}
```
Using the code presented, the compiler returns the error: 
```
StaticJsonBuffer<200> jsonBuffer;

  JsonObject& device = jsonBuffer.createObject();
  device["id"] = 398274;
  device["device"] = "Ambient+";
  device["version"] = "1.2";

  JsonObject& data = device.createNestedObject("data");
    data.["temperature"] = "22.5";  
    data.["humidity"] = "83";
    data.["lux"] = "4208";
    data.["CO2"] = "465";

  //root.printTo(Serial);
  device.prettyPrintTo(Serial);
```
>
sketch_jul01a:19: error: expected unqualified-id before '[' token
expected unqualified-id before '[' token

bblanchon, 2015-07-01:
@markusand just remove the extra `.`, like

    data["temperature"] = "22.5";

markusand, 2015-07-01:
I don't even know how I missed that. What a stupid mistake. Thanks.

bblanchon, 2015-07-01:
@markusand, you're welcome :wink: 
Thanks for using ArduinoJson.