bblanchon/ArduinoJson #59: JsonParserExample fails on Nano

kontur, 2015-03-16:
Hey,

I've been testing the `JsonParserExample` and when testing it with the Arduino Nano board the Serial just does not print anything at all. Using the exact same code (the example code) and uploading to a regular Arduino Uno, the Serial Monitor shows the correct readings.

Is this a know issue, or do you have any idea how to further debug it?

bblanchon, 2015-03-16:
No, it's not a known issue.
It's quite a surprise because I'm regularly testing on an Arduino Duelmilanove which has the same ATmega328.
Did you test another sketch on the Arduino Nano?

kontur, 2015-03-16:
Yep, I had some sketch running on the Nano before, then tried the example using JSON and failed. So I then tried to run it on Arduino Uno and there it worked. Tried re-plugging Arduino, re-uploading. Anything else I can try to get more info on why it is just failing? I don't get any errors, just nothing in the Serial monitor.

bblanchon, 2015-03-17:
Please try to insert `Serial.println()` between each line to find where it fails exactly.
(also don't hesitate to add them in the ArduinoJson code itself)

kontur, 2015-03-18:
Hey, sorry for the delay. I tried adding in debug print statements, but the program doesn't even reach the first I added just after `Serial.begin(9600);` - my assumption is that something about the Nano make the included library crash/silently fail. Let me know if I can test anything else.

bblanchon, 2015-03-18:
You need to make sure that the `Serial.println()` works.
It's probably not related to ArduinoJson.
Please try other examples shipped with the Arduino IDE.

kontur, 2015-03-18:
I think there is more to it.

This "hello world" works and prints to Serial:
```
void setup() {
  Serial.begin(9600);
  Serial.println("hello world");
}

void loop() {
  Serial.println("hello world");
} 
```

But this "hello world" doesn't print:
```
// Copyright Benoit Blanchon 2014
// MIT License
//
// Arduino JSON library
// https://github.com/bblanchon/ArduinoJson

#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  Serial.println("hello world");

  StaticJsonBuffer<200> jsonBuffer;

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  const char* sensor = root["sensor"];
  long time = root["time"];
  double latitude = root["data"][0];
  double longitude = root["data"][1];

  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {
  // not used in this example
}
```

bblanchon, 2015-03-18:
Is it the ATmega328 or the ATmega168 version?
It may be due to the RAM usage, please try:

     StaticJsonBuffer<50> jsonBuffer;

It should be enough for the sample string.

kontur, 2015-03-18:
It's the 328 version. Great pointer here! A value of 50 gave me `parseObject() failed`, so I tried with 100 and that parses correctly :)

So it's a matter of adjusting the array size of the buffer to match the limitations of the board and requirements of the string to parse. I'll be able to work around that just with trial and error, I believe. Good to confirm that's the reason. Any other pointers for this?

I'll close the issue, as I consider it resolved :+1:

closed 2015-03-18 by kontur

bblanchon, 2015-03-18:
I'm glad it's working with 100 bytes, but there is definitely something wrong.
An Arduino Nano with an ATmega328 has 2KB of SRAM which is more than enough for 200 bytes!

Maybe you configured the Arduino IDE with the wrong board, or maybe you found a bug.
What version of Arduino IDE are you using?

Although it's unlikely, we can't ignore the fact that it could come from an hardware issue, do you have another Arduino Nano to repeat the test?

kontur, 2015-03-19:
Running version 1.0.6. I would not be surprised if it is a hardware problem. It is not an official Arduino Nano but a replica, and it has been problematic in other ways, for example by not having an automatic reset on uploading sketches.
Unfortunately I don't have any other Arduino Nano to test with for the time being. If I will, I shall revisit this. For now, thanks for taking the time to look into this.

MYeager1967, 2015-03-20:
Not sure how or where to properly ask so please forgive me for doing it
here. In the add command where you have ("key", data), how do I include a
text string value as data? Trying to use the variable name is entering true
as the data element.  The text only changes when I update the sketch but I
have it assigned to a variable so I don't have to chase through the sketch
to do it.

bblanchon, 2015-03-20:
@MYeager1967 see issue #55

MYeager1967, 2015-03-20:
That does the trick. Thank you. Is there an index of known issues or does
one just ask when they can't figure something out. I'm nesting key pairs
under a header key and probably not the most efficient way of doing it. I'm
sure it's been covered somewhere and would love to learn to use the library
correctly.

bblanchon, 2015-03-20:
@MYeager1967, you should find anything you need in the [Wiki](https://github.com/bblanchon/ArduinoJson/wiki)