bblanchon/ArduinoJson #60: Reading JSON using serial communication on Arduino

renno-bih, 2015-03-30:
This library is great, helps me sending JSON data from Arduino to Web. But I also want to send JSON data from web to Arduino to control some of devices.
At the moment I'm using serial interface for communication between Arduino and Node.js script. I managed to send JSON stream to Arduino but I cannot store it to variable that can be parsed using this library.

I tried this method to read each line:

```
 while (Serial.available() > 0)
    {
        char recieved = Serial.read();
        inData += recieved; 

        // Process message when new line character is recieved
        if (recieved == '\n')
        {
            Serial.print("Arduino Received: ");
            Serial.print(inData);

        }   
            inData = ""; // Clear recieved buffer
        }
    }
```

But this example doesn't work for reading line of JSON data. 
Any idea why?

bblanchon, 2015-03-30:
Maybe the end of line is not `\n`, it could be `\r` or `\0`.
Also, you should try `Serial.readBytesUntil()`

renno-bih, 2015-03-30:
Hmmm, not sure how to use Serial.readBytesUntil() with strings .... but I tried this:
```
if(Serial.available() > 0)
    {
        inData = Serial.readStringUntil('\n');
       Serial.println("data: "+inData);
       
       inData="";
     }
```

And strange thing is happening again ... 
For input:
```
{"type_json":1,"count_json":68,"codeValue_json":"20df40bf","bits_json":32}

```

I get this output:
```
data: {"type_json":1,"count_json":68,"codeValue_json":"20df40bf","bit

```

Any reason why it doesn't store the whole line of input text?

renno-bih, 2015-03-30:
I solved problem.
I was using delay function in my loop. Looks like delay function has influence on readStringUntil() ... 

Organizing my code in separate timers and leaving readStringUntil() in loop fixed my problem.

q2dg, 2015-04-03:
readStringUntil() has a timeout configurable with setTimeout()

closed 2015-04-07 by bblanchon

pilotak, 2015-11-05:
how do you parse your variable "inData" to JSON?