bblanchon/ArduinoJson #62: JSON generator casts bool to int

fmntf, 2015-04-07:
Hello,
I'm using ArduinoJson 4.2, parsing works good, but generation has something weird:

      StaticJsonBuffer<200> responseJsonBuffer;
      JsonObject& response = responseJsonBuffer.createObject();
      response["success"] = true;
      response["value"] = 123;

The generated JSON is:

    {"success":1,"value":123}

Why is the boolean casted to int?
Thank you!

bblanchon, 2015-04-07:
Hi,

I'm really surprised because [there is a unit test just for that](https://github.com/bblanchon/ArduinoJson/blob/master/test/JsonObject_PrintTo_Tests.cpp#L105).

The only explanation I see is a `#define` somewhere in the code, maybe in another library.
Please look for something like:

    #define true 1
    #define false 0

We used to write this kind of trick in the old C time.

Regards,
Benoit

fmntf, 2015-04-07:
Hi Benoit,
probably you are right. Other than <ArduinoJson.h> I'm including <adk.h>, which includes a lot of files in cascade. Grepping for "define true" in Arduino IDE directory I get some results (stdbool.h, system.h).
99% the issue comes from ADK.

Are you aware of any workaround when a library redefines true/false?

closed 2015-04-07 by fmntf

bblanchon, 2015-04-07:
Either

     #undef true
     #undef false

or

    response["success"] = (bool)true;

or

    response.add<bool>("success", true);

(caution, the last one may not work with ArduinoJson 5.0)

fmntf, 2015-04-07:
Thank you so much!