bblanchon/ArduinoJson #63: serialisation of JsonObject

jhoogeboom, 2015-04-16:
Excuse me if this is a silly question and I've just overlooked something, but the only way to serialise the generated JsonObject is to run it with PrintTo(); method right? Using the stream/print interfaces of arduino.

Is there a straightforward way to get the serialised data in a buffer? say a char array? The reason I ask is I would like to store some json on eeprom as configuration for a node as well as have nodes send json over RF using packets.

bblanchon, 2015-04-16:
char buffer[256];
    myJsonObject.printTo(buffer, sizeof(buffer));

See https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON#use-a-classic-char

jhoogeboom, 2015-04-16:
great! thanks

closed 2015-04-16 by jhoogeboom