bblanchon/ArduinoJson #66: JsonVariant jv(jobj) not allowed

firepick1, 2015-04-18:
Assume we have:
   JsonObject jobj = ...;
We can always type:
   JsonVariant jv;
   jv = jobj;
But we cannot type:
   JsonVariant jv=jobj;
This is peculiar to me since the two seem semantically identical. The compiler does not allow the second form and warns about:
   ReferenceType::ReferenceType(const ArduinoJson::Internals::ReferenceType&) is private.

bblanchon, 2015-04-18:
I know, that's the kind of issue I'm adressing with version 5.0.
I made some progress on [this branch](https://github.com/bblanchon/ArduinoJson/tree/jsonvariant-constructor) but I'm not yet satisfied with the result.

firepick1, 2015-04-18:
Thanks for looking at it. C++ is so gnarly and I was afraid to try fixing it--it needs your expertise. :D

bblanchon, 2015-04-18:
Normally, a user doesn't manipulate `JsonVariant` directly.
I'm quite curious, can I see a code snippet?

firepick1, 2015-04-18:
Benoit,

I am used to the Jansson style of a common class for Json nodes. For
Jansson it is json_t*. For ArduinoJson it is JsonVariant.

I am using ArduinoJson to implement something similar to the TinyG Json
protocol where nodes can have different types according to context. Coding
as JsonVariant and using is<type> makes generic code possible. Example is
attribute for x coordinate. All of these have meaning:

"X":""
"X":123
"X":123.45
"X":[1,2,2,3]

The first is a command that requests value of X. The others set X to a
value or a series of values for specifying CNC position.

ArduinoJson is handling the above beautifully (thank you!) and I just
bumped into a few quirks that are minor but noticeable.

😀

bblanchon, 2015-04-18:
Wow!
Generic programming... templates...
I can definitely tell that you're not the average Arduino user
:wink:

firepick1, 2015-04-18:
Your library is most excellent and is perfect for the job. Jansson is fine
for TI Sitara but your code hits the arduino sweet spot. I am so glad I
don't have to write your library from scratch!

I will send you an update once I am finished in a week or two. 😀

bblanchon, 2015-05-31:
Available in [v5.0 beta 1](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.0-beta-1)

closed 2015-05-31 by bblanchon

firepick1, 2015-05-31:
😄