bblanchon/ArduinoJson #67: Stack overflow when printing double

miloyip, 2015-04-21:
When printing large/small `double`s, using `%f` will generate very long output and generate stack overflow, as shown in the following debug section:

~~~
frame #4: 0x0000000100006361 nativejson_debug_x64_gmake`Print::print(this=0x00007fff5fbfe5c0, value=1.7976931348623157E+308, digits=20) + 129 at Print.cpp:24
   21  	size_t Print::print(double value, int digits) {
   22  	  char tmp[32];
   23  	  sprintf(tmp, "%.*f", digits, value);
-> 24  	  return print(tmp);
   25  	}
   26  	
   27  	size_t Print::print(long value) {

(lldb) p tmp
(char [32]) $0 = "179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.00000000000000000000
~~~

I think a suitable format should be `%.*g`, though the meaning of `digits` are not the same.

bblanchon, 2015-04-21:
It's tough because I have to emulate Arduino's behavior.
I'll try to find a workaround...

miloyip, 2015-04-22:
To emulate Arduino's behavior, I think you may refer to its [implementation](https://github.com/arduino/Arduino/blob/db8cbf24c99dc930b9ccff1a43d018c81f178535/hardware/arduino/sam/cores/arduino/Print.cpp#L214).
It simply print big numbers as `ovf`. Although it does not fulfill the JSON standard, I think it is fine for limited system.

bblanchon, 2015-04-23:
It's fixed.
I decided to switch to "%g" instead of using "ovf" like Arduino.

closed 2015-04-23 by bblanchon