bblanchon/ArduinoJson #68: Improve out-of-memory handling

firepick1, 2015-04-26:
Benoit, I made a few tweaks in my branch to handle several out-of-memory edge cases that were blowing up my code. You might be interested in this:
https://github.com/firepick1/ArduinoJson/commit/bc8c3d988e1f4a0fba3207ab25b5a086f2af36e8

bblanchon, 2015-04-26:
What happened before you added the changes?
It should be 100% safe to manipulate invalid `JsonVariant`, `JsonArray` and `JsonObject`, every operation should be nops.

bblanchon, 2015-04-26:
PS: Do you think you can write a unit test?

firepick1, 2015-04-26:
I've got a unit test in my code that fails without these changes. I'll try
to add one in ArduinoJson.

It was failing on something like this:
StaticJsonBuffer<JSON_OBJECT_SIZE(4)> jbtiny;
{"a":{"b":{"c":{"d":123}}}

jbtiny.parseObject() returns  success.
However, a printTo() of the parsed JSON yields:
{"a":{"b":{"c":{}}

I knew that your intent was to make the invariants mutable, but I couldn't
figure out why this particular case broke, so after getting a headache I
slapped a goto bandaid on it and it seemed to work.

bblanchon, 2015-04-26:
Excellent!
I'll fix this ASAP.

firepick1, 2015-04-26:
Cool. Thanks! I was trying to figure out the unit tests and that would take
some time.

bblanchon, 2015-04-26:
I fixed the bug in a completely different way.
Please have a look at [branch issue68](https://github.com/bblanchon/ArduinoJson/commits/issue68).
I'll merge into master ASAP.

firepick1, 2015-04-26:
Thanks, Benoit!

Your fix feels much better than my bandaid. :D

closed 2015-04-27 by bblanchon