bblanchon/ArduinoJson #70: jsonobject prints string as true

MornyDK, 2015-04-28:
Hi

I'm planing on using the 1sheeld (1sheeld.com) to send and receive SMS with the Arduino. The SMS is to be generated and handled in a .net application. I'm planing on using json in the communication between the .net application and the Arduino.
I have a problem in the following code, when making the println in the beginning the number and the text is printet as expected in the 1sheeld terminal shield for debugging purposes. When making the println og the buffer in the end, I'd expect something like this {"PhoneNo":"555890123","Message":"The text in the SMS"} as I'm adding 2 strings to the jsonobject.

Insted I'm getting this: {"PhoneNo":true,"Message":true} - Why??

/MornyDK

void receiveSms(String number ,String text)
{
  Terminal.println(number);
  Terminal.println(text);
  
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["PhoneNo"] = number;
    root["Message"] = text;
    char buffer[256];
    root.printTo(buffer, sizeof(buffer));
    Terminal.println(buffer);
    Terminal.println("");
}

bblanchon, 2015-04-28:
This is a know issue that should be fix in next version.
For now you need to write:

    root["PhoneNo"] = number.c_str();
    root["Message"] = text.c_str();

Sorry about that.

See also issue #56.

closed 2015-04-28 by bblanchon

MYeager1967, 2015-04-28:
Will fixing this issue break the work around? If it does, that’s fine. I’d just like to know before I grab the latest version and break a LOT of code…

 

Mike

bblanchon, 2015-04-29:
No it wont break this code.
It will just add a shortcut.