bblanchon/ArduinoJson #71: const char* operator

arielsegura, 2015-04-28:
I'm using this library on an arduino project. I need to parse a IP sent in a JSON Object so what I'd like to do is something like: 

const char* gateway = root["gateway"];
char* token = strtok (gateway ,"."); 
and so on...

Notice that this won't work because strtok needs 'char*' instead of 'const char*'. So my question is why that operator returns const char* instead of char*? Would it be possible to return char*? I'm willing to help if neccessary. 
Thanks in advance. 
Regards,
Ariel

bblanchon, 2015-04-29:
That's an excellent question Ariel.

Indeed when using the JSON parser, you pass a `char*` to `parseObject()` so why returning a `const char*`?

On the other hand, when using the JSON generator, you write statements like:

    root["hello"] = "world";

In that case, `"world"` is really a `const char*`, it would be wrong to cast it to a mutable `char*` if someone reads the value back.

arielsegura, 2015-05-06:
Sorry for the delay, a lot of work over here...

Why returning a const char*?
Because if I only write this: 

char* gateway = root["gateway"];

The compiler says " conversion from 'ArduinoJson::JsonVariant' to 'char*' is ambiguous " and it gives some alternatives (One of them is using const char). 

So what I had to do (But it's clearly not the best option) is:

const char* gatewayString = root["gateway"];
char octetoAux[20];
memset(octetoAux, 0, 20);
strcpy(octetoAux, gatewayString );
char* token = strtok (octetoAux,".");

Perhaps there is another approach that I can't see.

bblanchon, 2015-05-06:
The other approach is called "casting":

    const char* gatewayString = root["gateway"];
    char* octetoAux = const_cast<char*>(gatewayString);

closed 2015-05-23 by bblanchon