bblanchon/ArduinoJson #73: reference question

jhoogeboom, 2015-05-01:
I would like to use ArduinoJson in a configuration storing-retrieving library that can write a JSON-string to EEPROM and back again. It would make sense to read and deserialize the json from EEPROM so that other functions/objects can retrieve key-value pairs.

But you can't declare a reference and not initialise it:

```JsonObject& root;
root = jsonBuffer.parseObject(json);```

This is, if I understand correctly, by design so that you only keep things in memory for short times? So what could be done is to simply create a temporary root every lookup, but that seems pretty inefficient to do that for every lookup?

bblanchon, 2015-05-01:
This is an excellent idea :+1:

You're right, you can't define a reference and not initialize it.
But you can use a pointer instead:

    JsonObject* root;
    root = &jsonBuffer.parseObject(json);

Or, you could use a `JsonVariant`:

    JsonVariant root;
    root = jsonBuffer.parseObject(json);

Both workarounds have their drawbacks.

Can you be more specific about the API that your library will offer?
In particular, how a user will load, save the config, and how she will get or set a value.
It may inspire other solutions.

jhoogeboom, 2015-05-02:
Ahh of course, yes that makes sense. Still fleshing out what the functions and data inputs and outputs should be. Have you seen the 2.0 EEPROM arduino library? Would be interesting to just directly access the EEPROM as a writable char array. So far it only offers const char* which I think is not right for arduinojson since it doesn't allow writing.

bblanchon, 2015-05-05:
Writing to the EEPROM would be easy, we just have to create a implementation of `Print` for that purpose.

But parsing a JSON string stored in EEPROM is another story.
ArduinoJson need to modify the input string to insert `'\0'` and unescape special characters.
So it would require copying the JSON string into the RAM anyway.

jhoogeboom, 2015-05-05:
I'm trying to assess whether this EEPROM as pointer implementation might be valuable, it overloads the operators to write and read from the EEPROM to function as a bidirectional pointer. If we can point the json parser to traverse the eeprom directly, it would save a lot of memory, albeit a bit slower than ram.

```
EEPtr ptr = 10; //Create a pointer to 11th cell.

*ptr = 4; //dereference and write to EEPROM cell.

unsigned char val = *ptr; //dereference and read.

ptr++; //Move to next EEPROM cell.
```

bblanchon, 2015-05-05:
OK, but even if it works, the process of parsing the JSON string is destructive.

Let's say the JSON was:

    {\"key\":\"value\"}

After calling `parseObject()`, the memory will contain:

    {\"key\0:\"value\0}

It's not valid JSON anymore.

closed 2015-05-23 by bblanchon