bblanchon/ArduinoJson #74: Library not recognised by Sublime Text with Stino Plugin

max929292, 2015-05-04:
I am using sublime text with Stino plugin as my Arduino IDE. 

https://github.com/Robot-Will/Stino

It works great and correctly sees all my Arduino libraries except the ArduinoJson one.

This library is very important for my work. Please let me know how I can get it recognised in Stino.

Thank You

PS: I left the same issue on Stino as I do not really know in which area this falls.

bblanchon, 2015-05-04:
Thanks or reporting this issue.
I discovered Stino, which curiously I never heard about.
Great project!

I fixed the issue in [pull request 273](https://github.com/Robot-Will/Stino/pull/273)

closed 2015-05-04 by bblanchon

max929292, 2015-05-05:
Thanks for fixing and it works now!
Stino is great but unfortunately it doesn't work well together with the new Arduino IDE Versions (so I downgraded to 1.0.5 and it seems to work well).