bblanchon/ArduinoJson #75: Post request, need content length

max929292, 2015-05-05:
I am using POST requests to send data to a webserver. These requests only work if we add the *content length* in the request. I do not want to use a char conversion via printTo as then I need to define the size of the char array which I would like to keep dynamic.
I am directly pushing the json to my EthernetClient however I do not know if there is a simple way to pull the content length from it.

Would be great if you could give me some pointers.

EthernetClient client;
JsonObject& root = jsonBuffer.createObject();
....

    client.println("POST /datapoints HTTP/1.1");
    client.println("Host:  xxx.xxx.xxx.xxx:3000");
    client.println("User-Agent: Arduino/1.0");
    client.println("Accept: application/json");
    client.print("Content-Length: ");
    client.println(root.length);    //How can i get length of json? root.length() is not available
    client.println("Content-Type: application/json");
    client.println("Connection: close");
    client.println();
    root.printTo(client);

bblanchon, 2015-05-05:
I think the cleanest solution would be to call `printTo()` twice: the first time to count the bytes, and the second time to write the HTTP request.

To count the bytes, you'll need an implementation of `Print` that counts the written bytes.
Something like that (not tested):

    class ByteCounter : public Print
    {
     public:
        ByteCounter()
            : length(0)
        {
        }

        virtual size_t write(uint8_t c)
        {
            length++;
        }        

        int length() const
        {
            return length;
        }
      
      private:
        int length;
    };

And then, in your app:

    ByteCounter counter;
    root.printTo(counter);

    int contentLength = counter.length();

Please, give this a try.
If it works, we could integrate something like that in the library.

max929292, 2015-05-06:
I had to change the length variable into len as there was a conflict with length(). This one works for me:

    class ByteCounter : public Print
    {
     public:
        ByteCounter()
            : len(0)
        {
        }

    virtual size_t write(uint8_t c)
    {
        len++;
    }        

    int length() const
    {
        return len;
    }
    
      private:
        int len;
    };

I also noticed that the counter does not reset len. So if you instantiate it outside your loop and your json has length 10, it will go 10,20,30,40...

It would be really great if you could implement this into your library. The POST requests on Arduino only works with content length and so I think this could be a great feature for lots of people.

Thank you for such a nice library and responsive support!

SoundConception, 2015-05-09:
I'm also sending responses  with JSON encoded content, to HTTP requests received by my Arduino, so I agree it really great if this was implemented in the library too.

I've added a reset() to what you guys came up with, and it's working well in my codebase.

Thanks again for a great library. 

```Arduino
class ByteCounter : public Print
{
  public:
    ByteCounter()
      : len(0)
    {
    }

    virtual size_t write(uint8_t c) {
      len++;
    }        

    int getCount() const {
      return len;
    }

    void reset() {
      len = 0;
    }

  private:
    int len;
};
```

bblanchon, 2015-05-09:
I added `JsonObject::measureLength()` in v4.4 to implement that feature.

@max929292 and @SoundConception, I recommend you to upgrade because the final implementation is much smaller.

See by yourself:

    size_t measureLength() const {
      DummyPrint dp;
      return printTo(dp);
    }

    class DummyPrint : public Print {
     public:
      virtual size_t write(uint8_t) { return 1; }
    };

closed 2015-05-09 by bblanchon