bblanchon/ArduinoJson #76: PROGMEM support?

brunnels, 2015-05-11:
I'm pretty low on memory in a project so I'm trying to scrape a few together by moving strings into PROGMEM.  Would it be possible to add support for this or should it work and I'm doing it wrong?

```
// normal way
root["relays"].as<uint8_t>();
// using PROGMEM macro that doesn't work but does compile
root[(const char*)F("relays")].as<uint8_t>();
```

bblanchon, 2015-05-11:
It doesn't work with the current version of ArduinoJson.
Reading from Flash memory requires a bit of extra work: you need to call `pgm_read_byte()` etc.

I'll try to add that feature in 5.x but I don't make any promise yet because I fear that it will increase the code size significantly.

puppetmaster886, 2015-08-28:
I think maybe is easier to support PSTR() rather than F().

JimGaleForce, 2015-08-28:
in my sketch, I have a struct which has a type and a pointer... if the type is 'P' (or whatever), it reads the pointer thru the pgm_read_byte(), otherwise, normal string/int/bool/etc. this adds little overhead.

jarred-s, 2016-02-16:
In the meantime until this is implemented I have been using the following macro to reduce memory usage.

```
#define PSTR_READ_INTO(buffer_name, string_literal)		char buffer_name[sizeof(string_literal)]; strcpy_P(buffer_name, (char*)PSTR(string_literal))

PSTR_READ_INTO(relays, "relays");
PSTR_READ_INTO(another_one, "another_one");

root[relays].as<uint8_t>();
root[another_one].as<uint8_t>();
```

isaaccentenario, 2016-03-25:
hey friend, after a few prints to the arduino to send jsons to the serial port. I believe the memory fill that time. how can I do to solve this problem?

bblanchon, 2016-03-25:
@isaaccentenario who are you talking to and how is this related to the current issue?

isaaccentenario, 2016-03-25:
sorry, I commented in the wrong place :/.

q2dg, 2016-04-04:
Sorry if I'm a little offtopic, but I just wanted to note that in ARM Arduinos there's no PROGMEM stuff. Instead, we can use some libraries to do the trick (https://github.com/cmaglie/FlashStorage is one example). Maybe this should be taken into account.

bblanchon, 2016-10-09:
I just realize that this feature is partially supported since v5.0, thanks to an implicit conversion to `String`.

This is supported:

```c++
root[F("sensor")] = "gps";
```

This is **not** supported:

```c++
root[F("sensor")] = F("gps");
```

but can be made to work like this:

```c++
root[F("sensor")] = String(F("gps"));
```

However, **I do not recommend to use this** because it's terribly inefficient. Each flash string has to be first copied to the heap (this is done by the `String` constructor), then copied into the `JsonBuffer`.
I'll try to implement a direct copy to the `JsonBuffer`.

bblanchon, 2016-11-13:
Support for PROGMEM has been added in ArduinoJson 5.7.1.
Please note that the content of the flash string is copied into the `JsonBuffer` when needed (just like it's done for `String`), so the `JsonBuffer` will be bigger.

@q2dg, I took your comment into account: I concentrated the logic for handling strings in a very small area so that support for other types of string can be added easily in the future.
Have a look at commit 1ce16ce449e37235f6cb4c6d7d4fb7dda81ac17a to see how I added support for PROGMEM; it's a good start if you want to add another type of string.

closed 2016-11-13 by bblanchon