bblanchon/ArduinoJson #77: encoding JSON with char* giving some problems

gigi199596, 2015-05-22:
Hello,

So first, let me thank you for all this amazing work, the JSON is working great but I found something a little disturbing.

I use this in my JSON encoder:

char temp[25];
...
(timeSaver.TO_STRING_TIME()).toCharArray(temp, 25);
root["time"] = temp;

And I get this output in my JSON in the browser:
"time": "Èº\b";

Or when I print temp to the serial output, it gives me the correct value of "temp".
What am I doing wrong?
thx a lot for your help

gigi199596, 2015-05-22:
TO_STRING_TIME() returns me the actual time like "07:26:59" in the String format. I convert it to char* because a string in your JSON encoder gives "true" in the browser... Don't know why neither

bblanchon, 2015-05-22:
Your problem is probably related to the scope of the variable.

ArduinoJson never makes any copy, it just stores pointers.
It's an efficient approach but it only works if the pointer is valid at the time is needed.

Please try something along these lines:

    {
        StaticJsonBuffer<200> jsonBuffer;
        JsonObject& root = jsonBuffer.createObject();

        String time = timeSaver.TO_STRING_TIME();
        root["time"] = time.c_str();

        root.printTo(Serial);
    }

(I added the curly braces to emphasize the fact that the `time` variable must outlive the call to `printTo()`)

gigi199596, 2015-05-22:
Ca marche parfaitement ! Un tout grand merci pour votre rapidité de réponse :) Mon projet de prises contrôlées à distance voit le bout grâce à votre travail :dancers:

[image]

bblanchon, 2015-05-22:
Ravi de vous avoir rendu service :wink: 

Can you please add a link to your project in the wiki?
https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson

closed 2015-05-22 by bblanchon