bblanchon/ArduinoJson #79: Does ArduinoJson work with ATMega 32U4

grhabyt, 2015-06-05:
On two separate Arduino Yuns on two separate laptops, I can't get JsonParserExample to Run. I just get no output in the Serial Monitor. And if I change the lines
if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
to 
if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  } else {
    Serial.println("Success!");
  }

I still get nothing from the Serial Monitor. 

I've checked the Baud Rates. I've run Examples>04.Communication>SerialCallResponse and the Serial Monitor works fine there.

I'm starting to wonder if there is an issue with the memory architecture of the ATMega 32U4 (the brains in the Yun and the Arduino Leonardo). The example runs fine on several ATMega 328 Arduinos (Uno, etc).

Help?

JimGaleForce, 2015-06-05:
I have it working perfectly on a Leo. The Leo has two serials, so Serial. and Serial1. The Serial1 is the RX, TX while the Serial is the USB. Perhaps this is why you are seeing a difference?

grhabyt, 2015-06-07:
To clarify. I still can't get the Serial Monitor going when the ArduinoJson library is included on the Yun.  The good news is that the library is working and correctly parsing my json. So I'm debugging by changing blink rates on the pin13 LED. A bit silly, but the parsing is what matters and that works.

bblanchon, 2015-06-08:
Some Arduino's examples contains a special waiting loop for Leonardo.

See [SerialCallResponse.ino, line 37](https://github.com/arduino/Arduino/blob/9166828bfe36a04ae0089b32b5b0ac93ca3bf7ca/build/shared/examples/04.Communication/SerialCallResponse/SerialCallResponse.ino#L37), for instance:

     // start serial port at 9600 bps:
    Serial.begin(9600);
    while (!Serial) {
      ; // wait for serial port to connect. Needed for Leonardo only
    }

I'm not sure it really makes sense, but can you give it a try?

Please also try to comment lines from ArduinoJson's example until you manage to get something out of the serial port.

Lastly, although it's very unlikely, it could come from a hardware problem. It happened in issue #59:

> I would not be surprised if it is a hardware problem. It is not an official Arduino Nano but a replica, and it has been problematic in other ways [...]

closed 2015-07-16 by bblanchon