bblanchon/ArduinoJson #83: Library with esp8266

stimpy76, 2015-06-24:
Hello,

I have used you library with esp8266 it is working great only thing is that it take about
10sec before he have maked the JSON string and send data to server of client. Is
there something what i can do to make it quicker.

Greetings Michel

bblanchon, 2015-06-24:
Hello Michel,

I never got my hands on an ESP8266, so my answer will be quite generic.

First of all make sure you use `StaticJsonBuffer` and **not** `DynamicJsonBuffer`.
Then, verify the clock of the ESP8266, it may not be running at full speed.

As far as I can tell, the only bottleneck in `ArduinoJson` is the *O(n)* insertion in `JsonArray` and `JsonObject`, but it should only affect you if you have a big number of items.
This can be easily optimized by adding a pointer to the end of the linked list.

Can you show a bit of code?
Can you tell which function is slow?

Regards,
Benoit

stimpy76, 2015-06-24:
Hiya Benoit,

ESP8266 is running on 80Mhz what they say in de arduino ide and also can select 160Mhz but no 
difference.
Sure i wanna share some code this is what i use on this way.

JsonObject& prepareResponse(JsonBuffer& jsonBuffer) {
  JsonObject& root = jsonBuffer.createObject();
  float voltage = readvdd33() / 1000.0;
  root["voltage"] = voltage;
  sensors.requestTemperatures();
  float valuetemp = sensors.getTempCByIndex(0);
  root["temperature"] = valuetemp;
  root["alarm"] = analogRead(4);
  return root;
}

void writeResponse(WiFiClient& client, JsonObject& json) {
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  client.println();

  json.prettyPrintTo(client);
}

void setup() {
  // put your setup code here, to run once:
  // Define Pins
  pinMode(2, INPUT); 
  pinMode(4, INPUT);
  pinMode(13, OUTPUT);
  
  // Start Serial
  Serial.begin(115200);
  
  // Start Sensors
  sensors.begin();
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
 
  // Start the server
  server.begin();
  Serial.println("Server started");
  
  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();

  // Match the request
  int val;
  if (req.indexOf("/gpio/0") != -1)
    val = 0;
  else if (req.indexOf("/gpio/1") != -1)
    val = 1;
  else {
    StaticJsonBuffer<110> jsonBuffer;
    JsonObject& json = prepareResponse(jsonBuffer);
    writeResponse(client, json);
    //Serial.println("invalid request");
    client.stop();
    return;
  }
    
  digitalWrite(13, val);
  
  client.flush();
  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

Greetings,
Michel

bblanchon, 2015-06-25:
I tryed you're program on an Arduino Duelmilanove + Ethernet shield, running at 8 MHz.
I just replaced the sensor readings by static values and Wifi stuffs by Ethernet equivalents.
It's as fast as it can be, the results are instantaneous.

Please try static values in your program to be sure that the sensors are not responsible for the delay.
I also wonder if `readStringUntil()` is waiting for a timeout (see [manual](http://www.arduino.cc/en/Reference/StreamReadStringUntil)).
Then it could also be the [naggle algorithm](https://en.wikipedia.org/wiki/Nagle%27s_algorithm), it may need to be disabled.

closed 2015-06-25 by stimpy76

bblanchon, 2015-06-25:
@stimpy76, did you solve you're problem?

stimpy76, 2015-06-26:
Yeah i have solved problem it''s browser dependebal in mozzila take some time google chrome is faster and in android it is working fast really strange but ok it''s working thx for the helping.