bblanchon/ArduinoJson #84: Adding keys to object in a loop

SeanDS, 2015-06-25:
I am having trouble adding keys to a JsonObject within a loop:

```c
#include <ArduinoJson.h>

// output pins
unsigned int outputPins[8] = {2, 3, 5, 6, 7, 8, 9, 11};

// number of defined pins in list
const int NUMBER_OF_OUTPUTS = (sizeof(outputPins) / sizeof(int));

void setup()
{
  // open serial communications and wait for port to open:
  Serial.begin(9600);
  
  StaticJsonBuffer<1024> jsonBuffer;
  
  JsonObject& root = jsonBuffer.createObject();
  
  for (int i = 0; i < NUMBER_OF_OUTPUTS; i++) {   
    String key = String("pin_") + String(i);

    Serial.print("Key: ");
    Serial.print(key);

    String message = String(i);

    Serial.print(", value: ");
    Serial.println(message);
    
    root.set(key, i);
  }

  root.printTo(Serial);
}
```

This outputs on the serial terminal:

```
Key: pin_1, value: 1
Key: pin_2, value: 2
Key: pin_3, value: 3
Key: pin_4, value: 4
Key: pin_5, value: 5
Key: pin_6, value: 6
Key: pin_7, value: 7
{"pin_7":7}
```

So only the last iteration of the loop is added to the object. I guess the same value is being overwritten in each iteration, but I can't figure out why.

I'm using v5.0beta1. I know you don't encourage the use of String() but I tried to figure out a way to do this with chars and it was even less successful (meaningless garbage was printed, mostly).

Any help would be greatly appreciated! This is an excellent tool - thanks a lot for sharing it.

bblanchon, 2015-06-26:
Hi Sean,

This is a tough one. Several users felt into this trap (see #10, #57).

The reason is that, for performance reasons, ArduinoJson **never makes copies** of the strings.
In your program, you're reusing the same memory location to store all the keys, so ArduinoJson sees it at the same key.

It may look strange if you come from Java or C#, but C++ works that way.
In your program the variable `key` it deleted at the end of each loop iteration, and it is reused at the next loop iteration.

The solution is to allocate the memory outside of the loop.

Here is a suggestion:

	const char* keys[] = {
		"pin_0",
		"pin_1",
		"pin_2",
		"pin_3",
		// up to NUMBER_OF_OUTPUTS
	};

	for (int i = 0; i < NUMBER_OF_OUTPUTS; i++) {   
		root.set(keys[i], i);
	}

You should also consider using a `JsonArray`, as it seems more appropriate here.

Thanks for using the beta version, this is really helping me :wink:

Regards,
Benoit

SeanDS, 2015-06-26:
That worked - thanks very much! This has indeed exposed my lack of C++ exposure.

May I suggest that the wiki contains a small warning about this?

So far everything in 5.0beta1 is working as expected - good job!

bblanchon, 2015-06-26:
I added section [6. Do not assume that strings are copied](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#6-do-not-assume-that-strings-are-copied).
Don't hesitate to improve the wiki.

closed 2015-06-26 by bblanchon

SeanDS, 2015-06-27:
Thanks, will do!