bblanchon/ArduinoJson #85: How do I generate this?

probonopd, 2015-07-05:
How would I generate this?

```
[  
   {  
      "error":{  
         "address":"",
         "description":"link button not pressed",
         "type":"101"
      }
   }
]
```

bblanchon, 2015-07-05:
DynamicJsonBuffer jsonBuffer;
	JsonArray& root = jsonBuffer.createArray();
	JsonObject& error = root.createNestedObject().createNestedObject("error");
	error["address"] = "";
	error["description"] = "link button not pressed";
	error["type"] = "101";
	root.prettyPrintTo(Serial);

will generate 

	[
	  {
	    "error": {
	      "address": "",
	      "description": "link button not pressed",
	      "type": "101"
	    }
	  }
	]

Don't forget to replace the `DynamicJsonBuffer` by a `StaticJsonBuffer` if you need to run on an embedded platform like an Arduino.

probonopd, 2015-07-05:
Thank you for the explanation. Maybe you'd like to add it to the examples.

bblanchon, 2015-07-05:
[Example added to the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Examples).

closed 2015-07-05 by bblanchon

pydesai, 2016-12-29:
How would I generate this?

{
  "timestamp" : 1483036273819,
  "values" : [ {
    "objectId" : 4,
    "datatype" : "int",
    "value" : "2"
  } ]
}