bblanchon/ArduinoJson #86: Smaller footprint for Arduino Pro Mini

entozoon, 2015-07-09:
Hi, sorry to be a pain as this library is great.
I was wondering if it would be possible to strip out any features to make it even smaller?

I only actually need to decode simple strings from json, not encoding or anything else fancy...
```
const int BUFFER_SIZE = JSON_OBJECT_SIZE(6);
StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

JsonObject& root = jsonBuffer.parseObject(serialBuffer);
```

On my arduino it adds about 3,654 bytes to my sketch size, which I know is barely anything.. but I'm frustratingly hitting memory limits.

Thanks :)

bblanchon, 2015-07-09:
Encoder and decoder have a lot of code in common, it would be very hard to split them.
I added the encoding feature in version 2.0, you can try with version 1.2, but the interface was very different.

Can you run `avr-nm` to see what is eating the space?
This article explains how to use it: https://blog.benoitblanchon.fr/arduino-code-size-and-assembly/

entozoon, 2015-07-09:
Oh wow, that's a brilliant little article! Eye opening for me, truly.

The majority of my memory is being taken up by the UTFT.h library I'm using (sigh!). As for ArduinoJson, it has:

```
0000017c T ArduinoJson::Internals::JsonParser::parseAnythingTo(ArduinoJson::JsonVariant&)
000000aa T ArduinoJson::Internals::JsonParser::parseObject()
0000007c T ArduinoJson::Internals::JsonParser::parseArray()
00000064 T ArduinoJson::Internals::QuotedString::extractFrom(char*, char**)
0000004a T ArduinoJson::Internals::JsonParser::skip(char)
0000004a T ArduinoJson::JsonObject::operator[](char const*)
...
```

Which I dare say are all quite important!
Thank you for your help, I'm going to go digging around in other libraries and strim things down. Super useful to know how efficient ArduinoJson is.

closed 2015-07-09 by entozoon

bblanchon, 2015-07-09:
There are also some size improvements in version 5.0.