bblanchon/ArduinoJson #89: ArduinoJson Parsing OpenWeatherMap Data

ThoEngel, 2015-07-25:
The parsing of the openweatherMap Json data does not work. The check "root.success()" fails.
Do you have any idea. Where is my mistake?

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

const char* ssid     = "SSID";
const char* password = "PW";

const char host[] = "openweathermap.org";

String data = "";   // String with json data
float temp = 0.0;

WiFiClient client;

unsigned long lastConnectionTime = 0;            // last time you connected to the server, in milliseconds
const unsigned long postingInterval = 10L * 1000L; // delay between updates, in milliseconds

void setup() {
  Serial.begin(9600);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // if ten seconds have passed since your last connection,then connect again and send data:
  if (millis() - lastConnectionTime > postingInterval) {
    httpRequest();

    char buffer[data.length() + 1];
    data.toCharArray(buffer, sizeof(buffer));     // Copy json String to char array
    buffer[data.length() + 1] = '\0';
  
    parseJSON(buffer);
  }

  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    data = client.readStringUntil('\r');
    //Serial.print(data);
  }
}

void httpRequest() {
  // close any connection before send a new request.
  // This will free the socket on the WiFi shield
  client.stop();
  delay(100);

  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  Serial.print("Requesting URL: ");
  // This will send the request to the server
  client.print(String("GET ") + "/data/2.5/weather?id=2906088&units=metric&cnt=1" + " HTTP/1.1\r\n");
  client.print("Host: api.openweathermap.org\r\n");
  client.print("Connection: close\r\n\r\n");
  // note the time that the connection was made:

  lastConnectionTime = millis();
  delay(10);
}

void parseJSON(char json[300])
{
  StaticJsonBuffer<300> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);
  if (!root.success())
  {
    Serial.print("ParseObject() failed");
    //return;
  }

  double temp = root["temp"];
  Serial.println(temp);
}

bblanchon, 2015-07-25:
This is probably due to the size of the `StaticJsonBuffer`, you probably have to increase it.

Also, if you use ArduinoJson v5.0-beta, you can avoid the `String` copy because `JsonBuffer::parseObject()` can take a `String` as a parameter.

ThoEngel, 2015-07-27:
Yes, it works. Thank you for your quick help.

bblanchon, 2015-07-27:
You're welcome :smile:

Can I ask you a favor?
I'm sure many other programmers want to read the OpenWeatherMap data.
Could you please add a working example in the "Examples" section of the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Examples

Thanks!

ThoEngel, 2015-07-27:
How can I access to sub elements like: main\temp,main\pressure...

JSON Result form OpenWeatherMap:
{
  "coord": {
    "lon": 10.01,
    "lat": 50.89
  },
  "weather": [
    {
      "id": 804,
      "main": "Clouds",
      "description": "overcast clouds",
      "icon": "04d"
    }
  ],
  "base": "cmc stations",
  "main": {
    "temp": 15.76,
    "pressure": 1002,
    "humidity": 93,
    "temp_min": 14,
    "temp_max": 17.22
  },
  "wind": {
    "speed": 3.6,
    "deg": 180
  },
  "rain": {
    "3h": 0.87
  },
  "clouds": {
    "all": 90
  },
  "dt": 1437978195,
  "sys": {
    "type": 3,
    "id": 4954,
    "message": 0.0097,
    "country": "DE",
    "sunrise": 1437968316,
    "sunset": 1438024406
  },
  "id": 2906088,
  "name": "Heringen",
  "cod": 200
}

ThoEngel, 2015-07-27:
But in  the moment I have no example that works. 
The access to root elements (like "name" "cod" or "id") works. 
How can I access to sub elements like main/temp oder main/presure?
If the access to sub elements works, I will add this example to your example section of the wiki. 
Thanks.

bblanchon, 2015-07-27:
double temp = root["main"]["temp"];
    long pressure = root["main"]["pressure"];

delcomp, 2015-07-27:
I have a simular problem in another project. Referring to above JSON table:
How to access  the "description" key in the table in the "weather" key?

delcomp, 2015-07-27:
Got it: 
data = root["weather"][0]["description"]

acasadoalonso, 2015-07-27:
I have also a working program for openweathermap, I will add it to the wiki

AC/.

closed 2015-07-31 by bblanchon