bblanchon/ArduinoJson #90: Longer then long variables

delcomp, 2015-07-27:
{"id":640794306199020323}

Following does not work because long is 32-bit.
long test = root["id"] ;

Actually I need 'id' as a string. So I tried the following:
const char* test = root["id"].asString();
But that returns an empty string.

Any ideas?

bblanchon, 2015-07-27:
I'm sorry, ArduinoJson doesn't provide any solution to this problem right now.

However I planned to rewrite the number parser for version 5.1.
I'll see if it's possible to make a "fall-back to string" behavior when there are too many digits.

delcomp, 2015-07-27:
Thanks.

I tried to use a double. But the lib returns 0.00? Any reason why this doesn't work:
double test = root["id"]

bblanchon, 2015-07-27:
Each value is stored in a `JsonVariant`, that can be either a string, a long or a double (to simplify).
The parser choose the type from the type in the json string:

     {"id":640794306199020323.0} // id is a double

     {"id":"640794306199020323"} // id is a string

     {"id":640794306199020323} // id should be long but parsing fails

Reading as a `double` any other type will return the default value of `0.0`.
You can check is a variant is a double by calling `JsonVariant::is<double>()`.

bblanchon, 2015-08-10:
Feature added to [version 5.0 beta 5](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.0-beta-5).

JSON token are now stored as `char*` and converted to the target type only when needed.

The following program will work as expected:

```c++
char json[] = "{\"id\":640794306199020323}";
JsonObject& root = jsonBuffer.parseObject(json);
const char* id = root["id"];
```

closed 2015-08-10 by bblanchon