bblanchon/ArduinoJson #92: Library 'hangs' while parsing

delcomp, 2015-07-27:
My app reads a JSON string every 10 seconds. Every appr. 5 minutes or so the app 'hangs'. Unfortunately this is in the json library.

This is my debug code:
              Serial.println("pre-json");
              JsonObject &root = jsonBuffer.parseObject(sTeslaResponse);
              Serial.println("post-json");

This the debug window:
Response: {"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":null,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"\u003Cinvalid\u003E","battery_range":135.25,"est_battery_range":100.81,"ideal_battery_range":107.51,"battery_level":43,"usable_battery_level":43,"battery_current":-0.1,"charge_energy_added":0.0,"charge_miles_added_rated":0.0,"charge_miles_added_ideal":0.0,"charger_voltage":0,"charger_pilot_current":0,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"motorized_charge_port":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":true,"eu_vehicle":true,"charger_phases":null}}
pre-json

As you notice 'pre-json' is output to the debug window, 'post-json' isn't. 

Any idea why the program might hang in your library?

bblanchon, 2015-07-27:
Usually this happens when the `JsonBuffer` is reused multiple times.
https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-staticjsonbuffer

But anyway, it should return an error and not hang, so I'll investigate...
What version are you using ?

delcomp, 2015-07-27:
V4.5
I use 'DynamicJsonBuffer jsonBuffer', not the static buffer.

The module is CC3200 based, it's not an Atmel CPU. It has 256kByte of SRAM, plenty of space for buffers. It's compatible with the Arduino IDE.

bblanchon, 2015-07-27:
The same issue can happen with a `DynamicJsonBuffer`.
You need to have the `DynamicJsonBuffer` in a local scope so it's discarded as soon as possible.
Do not use a global variable.

bblanchon, 2015-08-01:
Fixed in [version 4.6](https://github.com/bblanchon/ArduinoJson/releases/tag/v4.6)

closed 2015-08-01 by bblanchon

kabsila, 2015-08-02:
fatal error: src/DynamicJsonBuffer.cpp: No such file or directory
 #include "src/DynamicJsonBuffer.cpp"

bblanchon, 2015-08-02:
:cold_sweat:
Fixed in [version 4.6.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v4.6.1)