bblanchon/ArduinoJson #94: method to determine size in char

shirish47, 2015-07-31:
Do we have a method to determine size in char or byte of an JSON object? 
I want to know size of object to store it in bytes because later I am converting to base64 and I am not able to convert back from that if I preassume the buffer size of char.

bblanchon, 2015-07-31:
If you want to know the length of the JSON string, call:

    JsonObject::measureLength();

If you want to know the number of elements in a object, call:

    JsonObject::size();

If you want to known the size in bytes of an object with `n` elements call:

    JSON_OBJECT_SIZE(n)

And therefore, you can know the size currently used by an instance with:

    JSON_OBJECT_SIZE(jsonObject.size())

Finally, if you need to known the amount of memory used in a `JsonBuffer` call:

    JsonBuffer::size()

shirish47, 2015-07-31:
ok fine how do I get JsonBuffer from JsonObject?  to do this JsonBUffer::size().

and I am trying to copy the JsonObject to a char but don't want to fix the size of char. That is my problem.

bblanchon, 2015-07-31:
You cannot get the `JsonBuffer` from the `JsonObject`.
You have to use the `JsonBuffer` that you used to create the `JsonObject`.

shirish47, 2015-07-31:
I am doing this,
code>>>>>>>>>
char buffer[jsonBuffer.size()];                                   //<<<<<<<<<<here is my problem 
  root.prettyPrintTo(buffer,jsonBuffer.size());
  
    Serial.println();
    Serial.println(buffer);
    root.prettyPrintTo(Serial);
<<<<<<<<<end of code 

I get this output...
output>>>>>>>>>>>>

{
    "event": "User Rating",
    "Properties": {
    
{
    "event": "User Rating",
    "Properties": {
        "distinct_id": "1213",
        "token": "cd5c95e36ce1dfbc31629501f843a625",
        "Device": "567"
    }
}

<<<<<<end of output.
obviously buffer size is less but I want a method to know exact buffer size needed so as not to fix the size of buffer.

bblanchon, 2015-07-31:
It's `root.measureLength()` that you're look for.

shirish47, 2015-07-31:
however that does not help me ... :( any way I can get a base64 from your prettyprintto()?

that does not count I think the comma and "{}" .. somehow buffer is not filled with everything in Json

bblanchon, 2015-07-31:
Sorry, I didn't see you called `prettyPrintTo()`.
In that case, you need to call the `root.measurePrettyLength()`.
It doesn't count the null terminator, so you need to add one byte.

closed 2015-08-09 by bblanchon