bblanchon/ArduinoJson #96: Problem casting values

martinpineyro, 2015-08-02:
Hi: 

I'm trying to decode a JSON reveived from a server which looks like this: {"error":"0","version":"21","programa":"[00:45,570,950,50]"}

As you can see the last item is an array containing 4 values. I'm parsing this string using "parseObject" and it seems to work fine since I don't get errors when I check the result through the .success() method. 

The problem appears when I try to retrieve the values. I tried to cast the values implicitly and explicitly but haven't succeeded in neither of them. My code looks like this: 

```
  String JSONrecibido = "";
  while (client.available()) {
    char c = client.read();
    JSONrecibido += c;
  }

  char json[JSONrecibido.length() + 1];
  JSONrecibido.toCharArray(json, JSONrecibido.length() + 1);
  //Serial.println(JSONrecibido);
  Serial.println(json);
  
  //this is what "Serial.println(json)" shows (and what I need to parse: 
  //{"error":"0","version":"21","programa":"[00:45,570,950,50]"}
 
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
 
  int error_server = root["error"];
  int version_server = root["version"];
  String hora_luz_server  = root["programa"][0].asString();
  int humedad_maxima_server = root["programa"][1];
  int humedad_minima_server = root["programa"][2];
  

  Serial.print("error ");
  Serial.println(error_server);
  Serial.print("version ");
  Serial.println(version_server);
  Serial.print("hora luz ");
  Serial.println(hora_luz_server);
  Serial.print("humedad maxima ");
  Serial.println(humedad_maxima_server);
  Serial.print("humedad minima ");
  Serial.println(humedad_minima_server);
```

The output I get is: 

```
{"error":"0","version":"21","programa":"[00:45,570,950,50]"}
error 0
version 0
hora luz 
humedad maxima 0
humedad minima 0
```

I red that "0" is the result you get when the actual value doesn't match the target type in the case of integers and "null" in Strings. That's exactly what I'm getting. 

Sould I cast  the values into another type? I have tried all the possible ones (long, double, etc) but still can't get this working. Any suggestions??

THAKNS!!

bblanchon, 2015-08-02:
Your progam is right, it's the input that is wrong :wink:

Take a close look at the JSON input:

	{
	  "error": "0",
	  "version": "21",
	  "programa": "[00:45,570,950,50]"
	}

It's only three strings!
Therefore, ArduinoJson will extract 3 strings:

    const char* error = root["error"];
    const char* version = root["version"];
    const char* programa  = root["programa"];

To convert the strings to integers, you can call `atoi()`, and to parse the complex string, you can use `strtok()`. 
But that's another story.

The program you wrote would have been perfect if the input was:

	{
	  "error": 0,
	  "version": 21,
	  "programa": ["00:45",570,950,50]
	}

martinpineyro, 2015-08-02:
Great!! Thanks for the quick response. I formatted correctly the JSON and everything works now.

closed 2015-08-02 by martinpineyro