bblanchon/ArduinoJson #97: Energia 16 MT

juanoboy, 2015-08-16:
Hi
I have been experimenting with the ArduinoJson library on Energia16 (and TI Launchpad cc3200), which have so far been successful.  However, Energia similar to Arduino IDE, has recently released a new multitasking version, which do provide an compiler error for the ArduinoJson libary.  Any chance that this will work in the future?
regards,

bblanchon, 2015-08-18:
I downloaded the new Energia 16 but I don't manage to build many of the examples.
Not only ArduinoJson examples but also the ones bundled with Energia.

For instance, `energia-0101E0016\examples\10.MultiTasking\ButtonEvent`has the following error:

```
In file included from Button.cpp:1:0:
Button.h:2:31: fatal error: xdc/runtime/Error.h: No such file or directory
 #include <xdc/runtime/Error.h>
                               ^
compilation terminated.
```

I think there is a problem with Energia itself.

juanoboy, 2015-08-20:
Thank you for looking into this issue.

I got the same error as above when in the menu Tools/Board selecting the LaunchPad w/ cc3200 (80MHz).  However, when selecting LaunchPad w/ cc3200 EMT (80MHz) the example above did compile.

Some of the included multitasking examples seems not to be finished, as dependent  on which board is selected, the example may or may not compile successfully.

bblanchon, 2015-09-13:
I added a package with the old layout.
It now works with Energia 16.
https://github.com/bblanchon/ArduinoJson/releases
https://github.com/bblanchon/ArduinoJson/releases/download/v5.0.2/ArduinoJson-v5.0.2-old-layout.zip

closed 2015-09-13 by bblanchon