bblanchon/ArduinoJson #98: error if string is incomplete?

domonetic, 2015-08-20:
Hello, I tested the API 4 with Example JsonParserExample
Several times if I try to decode errors introduced stops working.
I use it to send and receive orders for wifi, in which they can become incomplete .
What do I do wrong?

#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);

  StaticJsonBuffer<200> jsonBuffer;
  char* json[] = {"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":}",
                    "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.3]}",
                    "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080]}",
                    "{\"sensor\":\"gps\",\"data\":[48.756080,2.302038]}",
                    "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}" 
                  };
                    
  for(int f=0;f<5;f++)
  {
    JsonObject& root = jsonBuffer.parseObject(json[f]);
    if (!root.success()) 
    {
      Serial.print("parseObject(");
      Serial.print(f) ;  
      Serial.println(") failed");
    }
    else
    {
      Serial.print("parseObject(");
      Serial.print(f) ;  
      Serial.println(") success");
  //    const char* sensor = root["sensor"];
  //    long time = root["time"];
 //     double latitude = root["data"][0];
 //     double longitude = root["data"][1];
 //     Serial.println(sensor);
 //     Serial.println(time);
 //     Serial.println(latitude, 6);
 //     Serial.println(longitude, 6);
    }
   }
}

void loop() {
  // not used in this example
}

==========================================

Output Serial Monitor:

parseObject(0) failed
parseObject(1) success
parseObject(2) success
parseObject(3) success
parseObject(4) failed

bblanchon, 2015-08-20:
You're reusing the same `StaticJsonBuffer`. It's full after 4 loop iterations, that's why the parse fail.

The solution is to move the declaration of `jsonBuffer` inside the loop.

Also please have a look at the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer

domonetic, 2015-08-20:
thanks , this library is very useful.

closed 2015-08-20 by domonetic