bblanchon/ArduinoJson #99: Error compiling examples in v5.0.0

hemalchevli, 2015-08-24:
Hi, I'm getting the following error in all examples.

Arduino: 1.0.6 (Windows 7), Board: "Arduino Uno"
C:\arduino-1.0.6\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=106 -IC:\arduino-1.0.6\hardware\arduino\cores\arduino -IC:\arduino-1.0.6\hardware\arduino\variants\standard -IC:\arduino-1.0.6\libraries\ArduinoJson\src C:\Users\HEMALM~1.CHE\AppData\Local\Temp\build1125316506046315289.tmp\IndentedPrintExample.cpp -o C:\Users\HEMALM~1.CHE\AppData\Local\Temp\build1125316506046315289.tmp\IndentedPrintExample.cpp.o 

In file included from C:\arduino-1.0.6\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArray.ipp:10,
                 from C:\arduino-1.0.6\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArray.hpp:133,
                 from C:\arduino-1.0.6\libraries\ArduinoJson\src/../include/ArduinoJson.h:8,
                 from C:\arduino-1.0.6\libraries\ArduinoJson\src/ArduinoJson.h:13,
                 from IndentedPrintExample.ino:7:
C:\arduino-1.0.6\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArraySubscript.hpp:17: error: 'template<class TImpl> class ArduinoJson::JsonSubscriptBase' used without template parameters

bblanchon, 2015-08-24:
I fixed it in 5.0.1.
Thanks for your report.

closed 2015-08-24 by bblanchon

hemalchevli, 2015-08-25:
That's really awesome, Thank you Benoit ! :)