bblanchon/ArduinoJson #101: Reading parsed numbers fails on esp8266

JeffAshtonCT, 2015-08-26:
Adding debugging messages to the provided the parsing example:

```
// Copyright Benoit Blanchon 2014-2015
// MIT License
//
// Arduino JSON library
// https://github.com/bblanchon/ArduinoJson

#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);

  StaticJsonBuffer<200> jsonBuffer;

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  Serial.println("reading: sensor");
  const char* sensor = root["sensor"];

  Serial.println("reading: time");
  long time = root["time"];
  
  Serial.println("reading: latitude");
  double latitude = root["data"][0];

  Serial.println("reading: longitude");
  double longitude = root["data"][1];

  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {
  // not used in this example
}
```

produces the following:

```
reading: sensor
reading: time

Exception (29):
epc1=0x40204377 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffe9b50 end: 3ffe9e80 offset: 01a0

>>>stack>>>
3ffe9cf0:  3ffe8841 00000000 3ffe8d70 3ffe9eac  
3ffe9d00:  3ffe9df7 3ffe9d24 3ffe9e00 40202760  
3ffe9d10:  3ffe8d70 3ffe9d24 3ffe8d70 4020219c  
3ffe9d20:  3ffe88c0 3ffe9d20 3ffe9d2c 3ffe9d4c  
3ffe9d30:  00000000 3ffe9df0 00000000 00000002  
3ffe9d40:  3ffe9eac 3ffe9df7 3ffe8e60 3ffe9da4  
3ffe9d50:  00000000 3ffe9dfb 00000000 00000001  
3ffe9d60:  3ffe9eac 3ffe9e00 3ffe8e60 3ffe9d20  
3ffe9d70:  3ffe9d74 3ffe9d8c 00000000 00000001  
3ffe9d80:  00000000 3ffe9e10 00000000 00000000  
3ffe9d90:  00000000 00000001 00000000 3ffe9e1a  
3ffe9da0:  00000000 00000000 00000000 3ffe9e0b  
3ffe9db0:  00000000 00000005 3ffe9eac 3ffe9d6c  
3ffe9dc0:  3ffe8e60 00000000 00000000 00000000  
3ffe9dd0:  00000000 00000000 00000000 00000000  
3ffe9de0:  00000000 00000000 00000000 000000a0  
3ffe9df0:  736e6573 6700726f 74007370 00656d69  
3ffe9e00:  31353331 31343238 64003032 00617461  
3ffe9e10:  372e3834 38303635 2e320030 30323033  
3ffe9e20:  2c003833 30332e32 38333032 00007d5d  
3ffe9e30:  00000000 00000000 00000000 00000000  
3ffe9e40:  00000000 00000000 00000000 00000000  
3ffe9e50:  00000000 00000000 00000000 3ffe9eac  
3ffe9e60:  3fffdc20 00000000 3ffe9ea4 40201e2b  
3ffe9e70:  00000000 00000000 3ffe8e60 40100378  
<<<stack<<<
```

bblanchon, 2015-08-26:
Hi Jeff,

From your comment, I assume the following:

1. This happens with version 5
2. It happens even if the additional log is not present.
3. Its was working fine with version 4

I don't have a ESP8266, so I'll need a little help from your side.

I think that to could come from the invocation of `strtol`, [in JsonVariant.cpp, line 50](https://github.com/bblanchon/ArduinoJson/blob/master/src/JsonVariant.cpp#L50):

```c++
return strtol(_content.asString, NULL, 10);
```

[The documentation](http://www.cplusplus.com/reference/cstdlib/strtol/) says that it's allowed to used `NULL` for the second parameter, but maybe it's not supported by your implementation.

Please try to replace this line by:

```c++
char* dummy;
return strtol(_content.asString, &dummy, 10);
```

If it doesn't work, please insert log in this area.
You may also need a similar fix for `strtod`, in [JsonVariant.cpp, line 36](https://github.com/bblanchon/ArduinoJson/blob/master/src/JsonVariant.cpp#L36)

JeffAshtonCT, 2015-08-26:
Yeah, that was the issue.  Had to fix strtod was well.  I'll put together a pull request tonight for you.

JeffAshtonCT, 2015-08-26:
Here's probably the real source of the issue,  they're not checking the pointer, 

https://github.com/esp8266/Arduino/commit/f73457de0d6b703b96a809e48b2163f16aa367e8

JeffAshtonCT, 2015-08-26:
Yeah, that was definitely it.  Thanks for your help.

closed 2015-08-26 by JeffAshtonCT

Diaoul, 2015-10-11:
Is this fixed yet? Because I'm experiencing this bug.