bblanchon/ArduinoJson #108: Cannot parse char/string

terminet85, 2015-08-29:
Hello,
first thank you so much for this project :) I would like to ask you to check my code 'cause I don't understand what's his wrong with my code.
My goal is to keep the json string from an webserver and parse it with yours library. I've tried both with chars and strings (I red you add it from 5.0 release)

```
void GetSettings() {
  bool begin = false;
  readString = "";
  Serial.println("Connessione...");
  if (client.connect(server, 80)) {
    Serial.println("Connesso");
    client.println("GET /arduino/growstation.json HTTP/1.1");
    client.println("Host: localhost");
    client.println("Connection: close");
    client.println();
  }
  else {
    Serial.println("Connessione fallita");
  }
  //Serial.println("sono prima di available");
  while (client.connected() && !client.available()) delay(1);
  if (client.available()) {
    while (client.connected()) {
      char c = client.read();
      if (readString.length() < 100) {
        if (c == '{') {
          begin = true;
        }
        if (begin) readString += c;
        if (c == '}') {
          break;
        }
      }
    }
  }
  Serial.println();
  Serial.println("Disconesso");
  client.stop();

  Serial.println(readString);
  //output {\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}

  StaticJsonBuffer<200> jsonBuffer;
  //char json[] =
  //"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
 //that's works

  String newstring;
  newstring = '"' + readString + '"';
  //add double quotes to match json example

  Serial.println(newstring);
  //"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}"

//attemps to try to convert string to char, but string would be supported!
/*
  int str_len = newstring.length() + 1;
  char json[str_len];
  newstring.toCharArray(json, str_len);
  Serial.println(json);
*/

  //will works with json[]
  //JsonObject& root = jsonBuffer.parseObject(json);

  //another attempt to convert string to char...
  JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(newstring.c_str()));

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  const char* sensor = root["sensor"];
  long time = root["time"];
  double latitude = root["data"][0];
  double longitude = root["data"][1];

  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}
```

Both if I try to convert string to char or use directly the string I got:
parseObject() failed

It works only if I use json[]

I hope you could take a look to it.

Thank you :)

terminet85, 2015-08-29:
Okay my bad solved.
removed backslash and double quotes from the string...

bblanchon, 2015-08-29:
Glad you found it :-)

I strongly encourage you to use a `char[]` instead of a `String`.
When you use a `String` the `JsonBuffer` will make a copy of the string to be sure it can modify it and that it will stay in memory.

Since you solved your problem I close the issue.

closed 2015-08-29 by bblanchon