bblanchon/ArduinoJson #109: Problem with prettyprintto

patrickwyggle, 2015-08-30:
I am probably doing something wrong, but prettyprintTo doesn't print anything in the following:
          StaticJsonBuffer<200> jsonBuffer;
          JsonObject& root = jsonBuffer.parseObject(json);
          if(root.success()){
            root.prettyPrintTo(Serial);
            if(root.containsKey("newparam")){
              JsonObject& newparam = root["newparam"];

with json: 
{"newparam":{"startupdelay":10000}}

Thank you!

Cheers,

Patrick

patrickwyggle, 2015-08-31:
The problem was in the arduino ide. I used version 1.0.6. I switched to 1.6.5.r5, and it works now.
Cheers,

Patrick

bblanchon, 2015-08-31:
It's supposed to work.
What is the target platform?
Can you post the full program?

bblanchon, 2015-09-01:
I tested with Arduino 1.0.6 and it works.
Please reopen the issue if needed.

closed 2015-09-01 by bblanchon