bblanchon/ArduinoJson #110: Parse a struct

asetyde, 2015-08-31:
i want use your library to parse my struct : 

'struct memory {'
'	char systemversion[5] = "0.7";
'	int set = 000; '
'	int status = 000; '
'	float humidity = 0;'
'	float temperature = 0;'
'	int delta = 2 ;'
'	int light = 0; '
'	boolean isrtc = false; // v or f'
'	char ip[16];'
'	byte mac_Address[6] = { 0xBA, 0xBA, 0xBA, 0xBA, 0xBA, 0xBA };'
'	byte scheduling[7][96][1]; // 1 or 0'
'	boolean releinverted = false;'
'	boolean reversesummer = false;'
'	boolean statusmachine = false;'
'	const char* ssid = "Network Office";'
'	const char* password = "domolab8790";'
'	// imported'
'	char arduinoName[DIM_STR] = "NoName";'
'	byte debounceDelay = 200;  '
'	long int countWrite = 0;'
'	//system struct'
'	char nameChannel[NUMB_CHAN][DIM_STR];'
'	char groupChannel[NUMB_CHAN][DIM_STR];'
'	boolean statusChannel[NUMB_CHAN];'
'	byte logicChannel[NUMB_CHAN];'
'	byte idChannel[NUMB_CHAN] = { 0, 0};'
'	byte counterIdShutter = 200;'
'	//system memory for action'
'	byte countAction = 0; //save the numbers of action in memory'
'	byte channelId[150]; //use this value for connect this information with the channel number'
'	byte ipAction[150][4];'
'	byte typeAction[150];'
'	int timeAction[150]; //this value is in seconds'
'	byte numberAction[150]; //in case of shutter action, this is id'
'
'	//variable for check the simply step configuration
'	byte simplyConfig = 0; // 0 = for nothing step configuration
'	byte ipConfig[4] = { 0, 0, 0, 0 }; //save the ip in the first step for the simply configuration
'
'};

I can try parse in byte after use single son string variable but .. when i go to java ...

ok jason string data is reconstructed correctly ..but my parsing in c i think no , therefore i must use only json system to pass data .. 
you have advice ? 

no problem with variable but with array ? i must create nest cycle to write or better "add" every single value ? 
struct is not so heavy , 2640 bytes but some internal array are heavy to cycle ... 

Help :)

asetyde, 2015-09-01:
Ok now i try to explain : 

i want use your json to port some data between my esp to my java program and viceversa.
Therefore i do this :  
create my son -> use printTo to put it on char buffer , after to send via udp  , i convert into char byte , after this i send on my udp port to my java program

asetyde, 2015-09-01:
my questions are : 

exist a better method to do this ?
how can i measure heavy of my jason when i create ? (before putting on char[] buffer) ?
java son library is compatible with your ? i must to convert my byte[] array to char[] and afre encode all

asetyde, 2015-09-01:
it's a little bit clear ?

bblanchon, 2015-09-01:
You can avoid the allocation of the `char[]`, see this example which sends through UDP:
https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonUdpBeacon/JsonUdpBeacon.ino

You can know the size of the JSON string by calling `measureLength()`.

Any Java JSON library will do the job.

asetyde, 2015-09-01:
thx , i work on it .

asetyde, 2015-09-01:
i underline a "probably problem " , i try to compile but your example  work and my not .. after 1 h i found the mystery ... 
function encode or decode  that i create where i create root ... therefore your buildjson must stay before your loop function or my send message .. there are a reason of that ? normal function can be stay before or after call function

bblanchon, 2015-09-01:
You'll probably find the answer in the documentation:
https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls

asetyde, 2015-09-01:
I not find clearly but if be a common , it's good idea write on your wiki .

Thx for reply!

asetyde, 2015-09-02:
you 've an email ? i want submit to you my code , because on java , i 've some problems .. son is new word for me

bblanchon, 2015-09-02:
I'm sorry. This is not the right place for this.
You should ask for help in [stackoverflow.com](http://stackoverflow.com/)

MYeager1967, 2015-09-02:
I understand why you referred him to stack overflow but they're going to
eat him alive. Worst place I've ever seen for programming support for new
programmers. :-)

asetyde, 2015-09-02:
i’m newbby to java .. i’m programmed in past but 5 years of c  .. max c++ is not a good air for come back to java … i try ..

asetyde, 2015-09-02:
Main problem is unique … time .. i ve my job to go fast , if could i learn and experiment i’m happy .. if 

http://stackoverflow.com/questions/32359750/how-send-json-from-c-wiring-to-java-and-viceversa <http://stackoverflow.com/questions/32359750/how-send-json-from-c-wiring-to-java-and-viceversa>

asetyde, 2015-09-03:
At the end of all none library goal to encode this stream .. i have a question ... you try with other son library or only in your encode or decode ... ??

bblanchon, 2015-09-03:
As I understand you message, you're implying that ArduinoJson isn't tested properly.
**This is a very unfair statement.**

ArduinoJson is tested very carefully: it has [99.96% of code coverage](https://coveralls.io/github/bblanchon/ArduinoJson), it's been validated by [external test suites](https://github.com/miloyip/nativejson-benchmark) and is used in production in [many projects](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson), including [Microsoft IOT](http://ms-iot.github.io/content/en-US/win10/SetupWVSA.htm).

Now, I'm closing this issue since it's not related to ArduinoJson anymore.

closed 2015-09-03 by bblanchon

acasadoalonso, 2015-09-03:
Benoit
I fully concur with you, it is very unfair
It is a wonderful piece of software and we deeply appreciate your generosity 

AC/.

asetyde, 2015-09-05:
It's not a fateful comment to Benoit , there are problem of  compatibility between esp and stand arduino therefore this library not work all good . I' ve been misunderstood .. :-1: