bblanchon/ArduinoJson #112: Teensy 3.1

venusdharan, 2015-09-01:
i downloaded the zip and added to teensy libraries folder opened the example to print JSON but it compiles without error but not printing anything to the serial monitor

bblanchon, 2015-09-01:
Does it work with ArduinoJson 4?
Did you configure the serial monitor to 9600 bauds?

venusdharan, 2015-09-01:
yes, i tried to fix it by removing pre-processors for arduino, then the follwing error came.
In file included from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\ArduinoJson.cpp:11:0:
C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/Arduino/Print.cpp:14:8: error: redefinition of 'size_t Print::print(const char*)'
size_t Print::print(const char s[]) {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:29:0,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/Arduino/Print.cpp:9,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\ArduinoJson.cpp:11:
C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\cores\teensy3/Print.h:59:9: error: 'size_t Print::print(const char*)' previously defined here
size_t print(const char s[])   { return write(s); }
^
In file included from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\ArduinoJson.cpp:11:0:
C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/Arduino/Print.cpp:22:8: error: redefinition of 'size_t Print::print(double, int)'
size_t Print::print(double value, int digits) {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:29:0,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/Arduino/Print.cpp:9,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\ArduinoJson.cpp:11:
C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\cores\teensy3/Print.h:74:9: error: 'size_t Print::print(double, int)' previously defined here
size_t print(double n, int digits = 2)  { return printFloat(n, digits); }
^
Error compiling.

bblanchon, 2015-09-01:
Version 5.0.2 contains a fix for issue #104 which was a problem in memory alignment.
Can you please test with your Teensy?
What about version 4, did you try?

venusdharan, 2015-09-02:
yes with v4
i got this error!
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonWriter.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:10,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: redefinition of 'class Print'
class Print {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/IndentedPrint.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: previous definition of 'class Print'
class Print {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/QuotedString.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonWriter.hpp:10,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:10,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: redefinition of 'class Print'
class Print {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/IndentedPrint.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: previous definition of 'class Print'
class Print {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/StringBuilder.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:12,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: redefinition of 'class Print'
class Print {
^
In file included from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/IndentedPrint.hpp:9:0,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\Internals/JsonPrintable.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: previous definition of 'class Print'
class Print {
^
In file included from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\cores\teensy3/Stream.h:24:0,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\cores\teensy3/HardwareSerial.h:163,
[... cut ...]
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\jsonarray.hpp:9,
from c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson.h:7,
from C:\Users\VSD\Desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\ArduinoJson\src/ArduinoJson.h:13,
from JsonGeneratorExample.ino:7:
c:\users\vsd\desktop\arduino-1.6.5-r2\hardware\teensy\avr\libraries\arduinojson\include\arduinojson\arduino\print.hpp:13:7: error: previous definition of 'class Print'
class Print {
^
Error compiling.

bblanchon, 2015-09-21:
I ordered a Teensey 3.1, it should arrive soon.
What IDE or environment did you use?

venusdharan, 2015-09-21:
teensyduino

bblanchon, 2015-09-23:
I received it today.
I'll probably try this week end.

bblanchon, 2015-09-27:
I figured it out.

You just need to add a `delay(1000)` after the call to `Serial.begin()`.
The Teensy probably needs a few cycles to initialize the serial port.

Also, be sure to follow the instructions from the Teensy tutorials:

> After uploading this to your board, use Serial Monitor
> to view the message.  When Serial is selected from the
> Tools > USB Type menu, the correct serial port must be
> selected from the Tools > Serial Port AFTER Teensy is
> running this code.  Teensy only becomes a serial device
> while this code is running!  For non-Serial types,
> the Serial port is emulated, so no port needs to be
> selected.

And that's how you spend 28€ :neutral_face:

closed 2015-09-27 by bblanchon