bblanchon/ArduinoJson #113: Encoding error

asetyde, 2015-09-03:
i want underline this error , variable pass from c to java but array seems empty ... but code seems not has error .

please help 

(mind is struct )
`JsonObject& encodejson(JsonBuffer& jsonBuffer) {`
``
`	// must included #include <ArduinoJson.h>`
`	JsonObject& root = jsonBuffer.createObject();`
``
`	//root["systemversion"] = mind.systemversion;`
`	root["set"] = mind.set;`
`	root["status"] = mind.status;`
`	root["humidity"] = mind.humidity;`
`	root["temperature"] = mind.temperature;`
`	root["delta"] = mind.delta;`
`	root["light"] = mind.light;`
`	root["isrtc"] = mind.isrtc;`
`	root["releinverted"] = mind.releinverted;`
`	root["reversesummer"] = mind.reversesummer;`
`	root["statusmachine"] = mind.statusmachine;`
`	root["ssid"] ="fghfhfgh";`
`	root["password"] = "fgdfg";`
`	root["debounceDelay"] = mind.debounceDelay;`
`	root["countWrite"] = mind.countWrite;`
`	root["counterIdShutter"] = mind.counterIdShutter;`
`	root["countAction"] = mind.countAction;`
`	root["simplyConfig"] = mind.simplyConfig;`
``
`	`
``
`	JsonArray& mac_Address = root.createNestedArray("mac_Address"); {  // byte`
`		for (int i = 0; i < 6; i++) {`
`			int value = mind.mac_Address[i];`
`			mac_Address..add(value);`
`		}`
`	}`
``

java code :

bblanchon, 2015-09-03:
Can you print the output of ArduinoJson, so we can validate that the array is present?

Also in your C++ code it's written `"mac_Address`, whereas in the Java code it's `"ip"`.

asetyde, 2015-09-03:
yes i correct , because i use old copy , no problem . thx 
i send my json to udp , you want stream of byte or ?

bblanchon, 2015-09-03:
```c++
root.printTo(Serial);
```

asetyde, 2015-09-03:
ok wait , also i found the problem , now al arrive but i change static to dynamic .. wait 
i

asetyde, 2015-09-03:
Ok problem is about Static and Dynamic allocation ... Dynamic all ok .
i try to send son , 303 bytes with dynamic and same with static but with static array not arrive .

asetyde, 2015-09-03:
But i think you must wait a board to  test , because esp is different from arduino in many thinks also  i use dynamic , output on serial  is correct.

bblanchon, 2015-09-03:
The `StaticJsonBuffer` was probably too small.

You can use the macros `JSON_ARRAY_SIZE()` and `JSON_OBJECT_SIZE()` to compute the required size at compile time.

In your case it could be:

    #define BUFFER_SIZE (JSON_OBJECT_SIZE(19)+JSON_ARRAY_SIZE(6))

because your object contains 19 values, and there is an array with 6 elements.

bblanchon, 2015-09-03:
PS: then if you use `String`s, you need to add the size of each because they'll be duplicated in the `StaticJsonBuffer`.

asetyde, 2015-09-03:
ok thx .

asetyde, 2015-09-05:
my jsonarrive to 1700~ bytes and now dynamic buffer not working , structure are corrupted . now i'll be back to jsonstatic to resolve .. i don't understand about string s that you 've said .

asetyde, 2015-09-05:
and at now i will try to escape from stack error on static buffer .. on <8000> ....4000 , 3000 is maximum ,  i 'm back to dynamic and split struct but it's not a good thing ...

closed 2015-09-13 by bblanchon