bblanchon/ArduinoJson #114: #include<ArduinoJson.h>  getting error of no such file

rickmoua, 2015-09-05:
I ran the example "JsonServer" which has the #include<ArduinoJson.h> and got this error below:
JsonServer.ino:7:25: fatal error. ArduinoJson.h: no such file or directory compilation terminated.

I download the version 5.0.2.zip from here https://github.com/bblanchon/ArduinoJson/releases
extracted and put in /sketchbook/libraries/ArduinoJson.

Not sure why getting the error. Any advise is appreciated.

bblanchon, 2015-09-06:
What version of Arduino IDE are you using?
Do the other example work?

bblanchon, 2015-09-09:
Is this on Energia 16, like #97 and #116 ?

bblanchon, 2015-09-13:
I added a package with the old layout (pre Arduino 1.5).
Please give it a try.

https://github.com/bblanchon/ArduinoJson/releases
https://github.com/bblanchon/ArduinoJson/releases/download/v5.0.2/ArduinoJson-v5.0.2-old-layout.zip

closed 2015-09-13 by bblanchon