bblanchon/ArduinoJson #115: Reuse JsonBuffer

naevtamarkus, 2015-09-06:
Hi,

I have to send some messages in a loop and, in order to save memory on the Arduino, I am trying to reuse a jsonBuffer. I do something like this:
```
  StaticJsonBuffer<200> jsonBuffer;
  while (X) {
      JsonObject& root = jsonBuffer.createObject();
      root["type"] = "sensor_record_data";
      JsonObject& data = root.createNestedObject("data");
      root["timestamp"] = ts;
      if (X) {
        data["sensor_1"] = data1;
        data["sensor_2"] = data2;
        root["ok"] = true;
      } else {
        root["ok"] = false;
      }
  }
```
This sends something like this (3 iterations):
```
Sending: {"type":"sensor_record_data","data":{"sensor_1":25,"sensor_2":37},"timestamp":1441519983,"ok":true}
Sending: {"type":"sensor_record_data","data":{"sensor_1":25,"sensor_2":35},"timestamp":1441520043}
Sending: {}
```

It seems like it gradually "degrades"... is there anything I can do to solve this? I guess you are going to ask me to create a separate jsonBuffer (inside a separate function, maybe) but that will take me more memory and I am really short. My plan is even use a global jsonBuffer that would be reused all the time...

BTW, I am using the latest 4.6.1

Thanks!

naevtamarkus, 2015-09-06:
Been diving through the code, but its C++ is too complicated for me to help with a PR. 
Maybe something like this inside JsonBuffer?:
```
public void clear(){ _firstNode = Null; }
```

asetyde, 2015-09-06:
You try with dynamic?

naevtamarkus, 2015-09-06:
> You try with dynamic?

Well, I thought dynamic was discouraged on memory usage issues

asetyde, 2015-09-06:
Also I 've problems because this library not support fully my device.
But if I do w Jsonbuffer with static i can transfer / storage with staticJsonBuffer<3000> about 400bytes with DynamicJsonBuffer about 500 ... Is not so good to trasfer and store structs and arrays but enought.

I think B. Use a system to calculate maximum space , you can try mod your code change static to dynamic and try with printTo.serial to see what is save and what is not

bblanchon, 2015-09-06:
@naevtamarkus, there is a very disturbing sentence in you question:

> in order to save memory on the Arduino, I am trying to reuse a jsonBuffer

I feel like you have a wrong understanding of how stack memory works.
Imaging that instead of a `StaticJsonBuffer`, you declare an `int`.
If you declare this `int` inside or outside of the loop, does it change the quantity of memory used in the loop? Of course not!
It's the same with `jsonBuffer`, it will use the same amount of memory.

Did you to try to declare `jsonBuffer` in the loop?
It's **exactly** the same as calling a `clear()` function at the end of the loop.

BTW, you are right to stick to static memory allocation, that is the biggest strength of this library.
Also, even if it wont solve this particular issue, you should give a try version 5.

naevtamarkus, 2015-09-07:
Hmmm I thought variable declaration was only memory allocation inside the stack frame, not initialization (unless you put ```int i = 1```) so, no, I did not try that.

What I meant by "in order to save memory" is that, in that function, I already use the JsonBuffer for other means so (does not show in my example) so, if I call another function with another JsonBuffer, I would be having two JsonBuffers in memory instead of one.

Anyway, I got around that problem by making the previous use of JsonBuffer a separate function:
- Main function does not have JsonBuffer
- It calls F1, who uses a JsonBuffer, finishes and frees that memory
- It makes the loop I described, and calls F2 inside N times
At the end is like replacing memory by computing cycles.

My question now (that I can't easily try what you suggested) is: is it worth reverting the changes (I have it in GIT) to try to declare the buffer inside the loop, see if it works and close this Issue? Is that going to work in your opinion? I don't see how, but again, I have not that good understand of C++ as you do.

And also, why are you suggesting version 5? Is there a ```clear()``` function there, or a way to mimic it?

Thanks!

bblanchon, 2015-09-07:
I suggested version 5, because it's better and smaller.
Please read [my blog post about it](https://blog.benoitblanchon.fr/arduino-json-v5-0/).

There is no `clear()` function and there will never be.
Here is why:

```c++
DynamicJsonBuffer jsonBuffer;
JsonObject& obj = jsonBuffer.createObject();
jsonBuffer.clear();
obj["key"] = "value";  // BOOOOOOM!!
```

In the code above, the reference `obj` is a pointer to memory in `jsonBuffer`; as soon as you call `clear()` this pointer is dangling and the program becomes unpredictable.
ArduinoJson has been designed to force you to use `jsonBuffer` in a scope so that this kind of nasty bug cannot appear.

Indeed, your workaround is exactly why I expected.
There is a good chance that this also leads to a better code, where responsibilities have been properly splitted between the main function and the serialization function.

closed 2015-09-13 by bblanchon

naevtamarkus, 2015-09-14:
Oh, crap, I forgot to answer... the BOOOOOM thing can happen at any time if you don't know what you are doing (aka languages and libs can't prevent you from doing stupid things)... like if you do ```jsonBuffer = null``` before, right? If features are documented it's user's responsibility to RTFM.

I managed to work around it by writing better (more convoluted and slower) code but I still thing it's a nice feature to have. But it's OK, you already made up your mind and that's not going to stop me from using this wonderful lib :)

bblanchon, 2015-09-14:
Pablo, it looks like you're more stubborn that me :wink: 

So you really want that `clear()` function?
Here is a secret: you can write it without modifying the library:

```c++
template<typename T>
void clear(T& instance)
{
    instance = T();
}
```

See issue #72 for explanations.
Remember that I don't recommend doing this and that I'll never add it to the library as explained above.

naevtamarkus, 2015-09-14:
Hehe, it's nice to see I am not alone in my stubbornness xDDD
Thanks for your time! It was very instructive :-)

Joypacharya, 2017-02-13:
But i want to flush json buffer.So, what i can do .please give me proper steps for that one.

 currently i sat 
StaticJsonBuffer<5000> jsonBuffer; in esp8266.
Also check in,
DynamicJsonBuffer jsonBuffer;
etc...
{
    etc...
}
jsonBuffer = 0; // use for reset buffer value.

So, tell me about this issue.
I got this type of message and reset my ESP8266.
Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff10:  13e28ce4 000002f4 3ffed0f0 401004f4  
3fffff20:  3ffee050 3ffee050 3ffed0f0 401072e8  
3fffff30:  00000000 063693f3 4020638b 402095ba  
3fffff40:  3ffee050 13e24394 40209a7a 00000008  
3fffff50:  402063f4 3ffe8c84 40209b27 3ffed1a4  
3fffff60:  3ffee050 0636aad3 3ffee050 60000600  
3fffff70:  4021b205 3ffed1a4 3ffee050 13e27627  
3fffff80:  4021b24a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee068 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 0000682a 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
ü

please help for this problem.