bblanchon/ArduinoJson #116: Energia16 - can't find the lib

smartrip, 2015-09-09:
Hi,

Thanks for the library, I really like the way it works - specially comparing with the other. 

Recently I switched to Texas Instrument CC3200 module (project requirement) and their IDE called Energia. Energia is similar to Arduino IDE as you probably know. I tried to integrate my code with your library, but wasn't successful. It works on regular Arduino board with Arduino IDE.

So, first I started to play with your examples, but couldn't even manage to make them work. Can you give me any hints?

The error message:
`JsonGeneratorExample.cpp:7:25: fatal error: ArduinoJson.h: No such file or directory
 #include <ArduinoJson.h>
                         ^
compilation terminated.`

My environment:
Windows 7, Energia 16, ArduinoJson v5.0.2

Fil

bblanchon, 2015-09-09:
Hi Fil,

This issue has already been submitted as #97.

I tried Energia myself and found that many bundled examples have the same problem.
So I assumed there may be something wrong with Energia 16, because, as far as I can tell, it was working properly with older versions.
I didn't have time to dig deeper into it.

Also, ArduinoJson 5 has [the new library layout](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification), this may be the reason why it's not working with Energia.
In that case, you may try ArduinoJson v4.6.1 because it has the old library layout.
If this is working, I can try to provide two kinds of package for ArduionoJson 5.x.

If it's still not working, one of us will have to submit the issue to the Energia team.

I'm closing this issue as a duplicate of #97, please continue the conversation there.

closed 2015-09-09 by bblanchon

smartrip, 2015-09-09:
Hi Benoît,

Very good lead! You are right. After quick investigation I found that the old version of the lib works fine (at least one of your examples). Would be cool if you could provide v5 of the lib in the old layout as a temp solution. I found two other libs that use the new layout and don't work with Energia (16). I will submit the issue with Energia team.

Thanks for your help.
F

bblanchon, 2015-09-09:
Hi Fil,

I'll do that I soon as I can.

Meanwhile, you can easily do a manual workaround by adding `ArduinoJson.cpp` and `ArduinoJson.h` like they used to be in version 4.
Have a look at [this commit that switched to the new layout](https://github.com/bblanchon/ArduinoJson/commit/283dffc0359b681b89ce403bca43c05fbfecc8b3)

smartrip, 2015-09-10:
Hi Benoît,

So I did a research and it turns that Energia lags a little bit behind the
original Arduino IDE development stream. They still didn't move into 1.5
code architecture, so introducing the new library layout is not feasible at
the moment. I think that the temp solution we were talking about is gonna
stay until Energia will make a move.

And thanks for the updates - I plan to migrate to your lib later this week.
I will let you know how did it go:)

Cheers,
F

Best regards

---

*Smartrip Technologies Limited*
Registered in Ireland, No.: 490049
Olympic House, Pleasants Street, Dublin 8, Ireland

T: +353 85 8800 117
M: info@smartrip.eu
W: www.smartrip.eu

*CONFIDENTIALITY : This e-mail and any attachments are confidential and may
be privileged. If you are not a named recipient, please notify the sender
immediately and do not disclose the contents to another person, use it for
any purpose or store or copy the information in any medium.*