bblanchon/ArduinoJson #118: String decoding

asetyde, 2015-09-14:
I found a strange thing , i not understand cleary why i must say explicitly that is a string .
arduinoname is a String type.

this is issue : 

[image]

[image]

with explicit asString , all ok : 
[image]

It's bug or what ? thx for reply

bblanchon, 2015-09-14:
You're right this is not the expected behavior.
I'll try to fix it, but I'm not sure it's even possible.

asetyde, 2015-09-14:
Thx , it's not a true problem , you 've included in wiki about casting .

bblanchon, 2015-12-22:
Please don't close it.
I'm still trying...

asetyde, 2016-10-28:
ook

bblanchon, 2016-11-24:
I don't think this is a solution for this.

To help new users, I added a section in `StringExample.ino` that describes this issue.

closed 2016-11-24 by bblanchon