bblanchon/ArduinoJson #119: Is it possible to parse Stream?

spotman, 2015-09-18:
Good evening, Benoît!

Thank You for this beautiful library!

I`m reading a JSON config string from EEPROM which may vary in length from 200 to 1500 bytes so reading it into temporary variable completely wastes the RAM. Is it possible to implement parsing JSON from Arduino Stream object?

PS: I may do it by myself if you`ll send me to the right direction.

bblanchon, 2015-09-18:
Good evening to you sir!
It's 3PM here, not really evenning :smile: 

There is a secret behind  zero-copy and zero-allocation features.
ArduinoJson relies on the fact that the data is already in memory.

Parsing the data from a `Stream` would require to dump the entire content in the RAM.
This feature could be added to `JsonBuffer`, but i'm not sure it's a good idea.

What do you think?
Is it acceptable that a `JsonBuffer` makes a copy of the stream's content?

spotman, 2015-09-18:
Oh, really 3PM, sorry! :)

The stream itself does not have entire content inside, it reads byte-by-byte from source, so it may perform data seek over 1MB file with near-zero RAM usage. I`ve planned to use this feature to save some RAM.

I thought that JSON parsing algorithm has something similar inside as it may parse every "layer" independently (linking it to current "parent" "layer" via pointer). After your explanation and some code digging I understood that you code has another algorithm. Well, there are two ways for me now:

- switch to MC with more RAM inside or
- write PR with alternative JSON parsing algorithm which is going layer-by-layer through stream-like interface.

I think the first way is cheaper today, but if you`re interested in improving parsing algorithm I`ll look forward the second way too :)

bblanchon, 2015-09-18:
I don't thinks this layer by layer approach makes sense.
`Stream` is a one way read source, you can't go backward.
This means that you can't do a multi-pass parsing, unless you copy the content in RAM.

spotman, 2015-09-18:
Why? JSON is a tree, parsing always starts from top level, every lower level may be "linked" to parent via pointer. When parser get `{` it creates new `jsonObject`, marks it as "current object" and parses everything till `}`. If parses reads another `{` it creates another `jsonObject`, sets its parent to previously "current object" and parses stream again till `}`. I think that it's possible to recreate tree structure without all the source string in the RAM.

Maybe I'm wrong, I need to perform some tests to check this approach :)

bblanchon, 2015-09-18:
It would still have to copy the *values* in RAM.
Your suggestion just saves the bytes for `{`,`}`,`,`,`[` and `]` which is relatively small.
So you would still have to copy the major part of the `Stream` in RAM.

spotman, 2015-09-18:
Yes. The current way is: EEPROM => string in RAM => jsonObject in RAM. It's memory usage formula is: `sizeof(string)` + `sizeof(jsonObject)`. So if JSON string has 1.5KB size then jsonObject would be ~1.5KB, but I have only 2KB RAM today.

I suggest to remove second step and connect parsing library to Stream source directly. So the memory usage formula would be: `jsonObject` + `Stream reading overhead`. It would have much lower memory usage summary, near original JSON string size. I need to do some tests to check the ability for byte-by-byte JSON parsing.

bblanchon, 2015-09-18:
It's not

    sizeof(jsonObject) + Stream reading overhead

at a minimum it's

    sizeof(jsonObject) + sizeof(JSON payload)

which is

    sizeof(jsonObject) + sizeof(string) - sizeof(ponctuation)

which is just a little smaller than the original.

spotman, 2015-09-18:
You are right if the algorithm would read the full source string into memory in one step. My suggestion is to read source string by "chunks" (key-value pair for jsonObject or single value for array) freeing memory after each "chunk" is processed and converted to data tree in RAM. So the largest memory usage would be: `largest source string "chunk"` + `sizeof(jsonObject)`.

PS: I hope I was correct in my explanation. I'm sorry for wasting your time...

bblanchon, 2015-09-18:
No worry, I like to hear people feedback... especially when it's challenging :wink: 

Can you give an example of a JSON input that we can try to process step-by-step?

spotman, 2015-09-18:
Well, let's start from this simple sequence:

```
{
  "byte": 255,
  "string": "value",
  "nestedIntArray": [ 0, 127, 255 ],
  "nestedObject": {
    "nestedInt": 255,
    "nestedString": "nestedValue"
  }
}
```

Algorithm of parsing is (skipping CR/LF/Space characters everywhere except string values between `"`):

- read first `{`, create root `jsonObject` object, mark it as 'current object' via pointer
- read first `byte` key till `:`, create property inside `current object`, mark it as `current key` via pointer
- read int value till `,` or `}` and assign it to `current key`
- read second `string` key till `:`, create property inside `current object`, mark it as `current key`
- read string value till `,` or `}` and assign it to `current key`
- read third `nestedArray` key till `:`, create property inside `current object`, mark it as `current key`
- detect `[`, create `jsonArray` object, set its parent to `current object`, assign it as value of `current key`, mark it as `current object`
- read till `,` or `}` and loop through every value inside of `nestedArray` adding them to `current object`
- read (if not) last `}`, set `current object` to parent of `current object` (root)
- set `current object` to root `jsonObject`, set `current key` to NULL,
- read fourth `nestedObject` key till `:`, create property inside `current object`, mark it as `current key`
- detect `{`, create `jsonObject` object, set its parent to `current object`, assign it as value of `current key`, mark it as `current object`
- read fifth `nestedInt` key till `:`, create property inside `current object`, mark it as `current key` via pointer
- read int value till `,` or `}` and assign it to `current key`
- read sixth `string` key till `:`, create property inside `current object`, mark it as `current key`
- read string value till `,` or `}` and assign it to `current key`
- set `current object` to root `jsonObject`, set `current key` to NULL,
- read (if not) last `}`, set `current object` to parent of `current object` (NULL) and exit

I do not see any issues with sequential parsing now. Do you see them?

bblanchon, 2015-09-19:
Here are very optimistic estimations of the required memory on an 8-bit AVR

1. 4 bytes for the `JsonObject`
2. 5 bytes for the string "byte"
3. 10 bytes for the `JsonObject` node
4. 7 bytes for the string "string"
5. 6 bytes for the string "value" + 10 bytes for the `JsonObject` node
6. 16 bytes for the string "nestedIntArray"
7. 4 bytes for the `JsonArray` + 10 bytes for the `JsonObject` node
8. 3*8 bytes of the `JsonArray`'s nodes
9. 0
10. 0
11. 13 bytes for the string "nestedObject"
12. 4 bytes for the `JsonObject` + 10 bytes for the `JsonObject` node
13. 10 bytes for the string "nestedInt"
14. 10 bytes for the `JsonObject` node
15. 13 bytes for the string "nestedString"
16. 12 bytes for the string "nestedValue" + 10 bytes for the `JsonObject` node
17. 0
18. 0

Total with this method: 178 bytes.

The original JSON string without the spaces is only 120 bytes.
(Of course I assume the JSON has been minified, it would be a complete nonsense to store or transmit "prettyfied" JSON strings on such memory constrained environments)

Total with original method: 206 bytes.
So that's only a 14% benefit, and this is very optimistic!

Now another problem arise: the size of the code will increase significantly.
The main struggle on a small MCU is usually not the RAM, but the Flash memory.

It's quite unusual (and a little crazy) to expect to parse 1.5KB of data when you only have 2KB of RAM.
Maybe you need a bigger MCU or to trim the useless data out of your JSON.
If your program gets bigger, you'll need a bigger MCU anyway.

spotman, 2015-09-21:
You are right, Benoît. My application needs MC with 4-8 KB of RAM and about 32 KB Flash. And my bottleneck is RAM, not Flash or CPU power. So the cheapest solution is to use another Arduino-compatible MC with a lot of RAM (like ATmega2560). Thanks for your patience and help!

closed 2015-09-21 by spotman

bblanchon, 2015-09-21:
Sorry I could not find a better solution.
Good luck with your project Denis.
Don't hesitate to come back later and add a comment to tell how the project went.

spotman, 2015-09-21:
Thanks again, I'll write results of my experiments here!

asetyde, 2015-09-21:
It's possibile yes 

Create json pass ti udp or trasformtobyte function 

You use another language ?

spotman, 2015-09-21:
@asetyde I'm sorry, but what do you mean?

asetyde, 2015-09-21:
You want parse into stream a Json ?

spotman, 2015-09-21:
@asetyde no, my JSON string is large (up to 1.5KB) and lays in EEPROM. So I want to read it sequentially via Stream and also parse sequenatially so no temporary variable with JSON string will be created in RAM. But today the current lib can not do this so I prefer to get MC with 4-8KB of RAM.

asetyde, 2015-09-21:
Ook ! I do misunderstanding ! Thx

bblanchon, 2017-01-13:
Built-in support for `Stream` has been added in [ArduinoJson 5.8.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.8.0).

spotman, 2017-01-13:
Thank you so much for that, @bblanchon ! It's opening the way to use the Arduino UNO for my project :) I'll give it a try next week.